autowarefoundation/autoware #2168: [Bug] When using `use_sukf` in imm_ukf_pda_track, naive_motion_predict does not work

k0suke-murakami, 2019-03-29:
## Bug report
When enabling `use_sukf` in `imm_ukf_pda_track`, `naive_motion_predict` does not properly predict objects.

### Required information:

- Operating system and version:
  - Ubuntu 16.04
- Autoware installation type:
  -  From source
- Autoware version or commit hash
  - 6a7d1b9f66fd353eb5c6ad8df942c433fff8e2a1
- ROS distribution and version:
  - kinetic
- ROS installation type:
  - from binaries
- Package or library, if applicable:
  -imm_ukf_pda_track 

### Description of the bug

SUKF model is supposed to treat all the objects as Constant Velocity model.
But, currently treats them as Random Motion model.

Currently, filling MotionModel in DetectedObject's `behavior_state` = 2.
But expecting `behavior_state` = 0 when enabling `use_sukf` because it treats objects as ConstantVelocity model.
https://github.com/autowarefoundation/autoware/blob/master/ros/src/computing/perception/detection/lidar_tracker/packages/lidar_imm_ukf_pda_track/nodes/imm_ukf_pda/imm_ukf_pda.cpp#L329-L343

`MotionModel`'s definition is here. 
https://github.com/autowarefoundation/autoware/blob/master/ros/src/computing/perception/detection/lidar_tracker/packages/lidar_imm_ukf_pda_track/include/ukf.h#L39-L44

### Steps to reproduce the bug
1. Launch `ray_ground_filter` in `sensing` tab
1. Launch `lidar_euclidean_cluster` in `computing` tab
1. Launch `lidar_imm_ukf_pda_track` in `computing` tab
1. In the terminal type `$  rostopic echo /detection/objects/objects[0]/behavior_state`
[image]

### Expected behavior
Get `bahevior_state` = 0

### Actual behavior

As I reproduced in above, get `bahvior_state` = 2

### Screenshots

### Additional information

gbiggs, 2019-03-31:
Please don't use screenshots of a terminal. Copy and paste the text instead.

k0suke-murakami, 2019-04-01:
> Please don't use screenshots of a terminal. Copy and paste the text instead.

Updated not to use a screenshot.

closed 2019-05-23 by amc-nu