autowarefoundation/autoware #2178: [Feature] TensorFlow - wrapper for TensorFlow C API

drwnz, 2019-04-01:
## Feature request

### Description
Implements a wrapper package for the TensorFlow C API, enabling easy use within ROS. It allows building a project using TensorFlow models for inference.

### Implementation considerations
The [TensorFlow C API](https://www.tensorflow.org/install/lang_c) is easy to install and does not require Bazel or many dependencies. However, the interface is not intuitive, hence the need for a wrapper package in ROS.

### Alternatives
-- The standard TensorFlow C++ API is built using Bazel which is difficult from within a CMake environment.
-- There is a [CMake version](https://github.com/tensorflow/tensorflow/tree/master/tensorflow/contrib/cmake) of the TensorFlow C++ API, however it is deprecated and not at the latest version of TensorFlow.
-- The Python API is easy to use but requires all nodes using TensorFlow models to be implemented in Python.

gbiggs, 2019-04-02:
There is support for `bazel` in `colcon`, although I don't know how well it works.

https://github.com/colcon/colcon-bazel

drwnz, 2019-04-02:
> There is support for `bazel` in `colcon`, although I don't know how well it works.
> 
> https://github.com/colcon/colcon-bazel

Thanks, I will check it out. However, the TensorFlow build for C++ is not that simple with a lot of dependencies and configuration depending on your system, whereas the pre-build C API can be downloaded and installed very easily either manually or from CMake, so I think it is a good option for minimal disruption in building Autoware.

wew84, 2019-05-19:
I built a package that you might find interesting that opens freeze graphs of segmentation / detection and publishes them to ROS. https://github.com/wew84/cnn_bridge

closed 2020-05-29 by gbiggs