autowarefoundation/autoware #2181: [Feature] autoware_msgs refactoring

alexandrx, 2019-04-01:
## Feature request

### Description
This feature was already proposed as PR #2066 and rejected purely on triviality. 
The PR was discussed and the procedure agreed with the reviewer.
The PR #2066 was issued and tested following due established procedures but was rejected by people following other procedures that did not correspond to the Autoware version.

Anyway, here is again as a feature request.

### Implementation considerations
It is already implemented and was already tested with `colcon`, in the PR description I mistakenly typed `catkin_make_release` instead of `colcon`.
New testing procedure was suggested in https://github.com/autowarefoundation/autoware/pull/2066#issuecomment-470405702, that can be done. 

### Alternatives
None

### Additional information
None

alexandrx, 2019-04-02:
@gbiggs Please explain why the title changes when there are plenty issues in 1.12 milestone either with `[Feature]` or `[Bug]` in the titles.

gbiggs, 2019-04-02:
@alexandrx We use labels to identify issue types now. They are easier to manage and search, and more visible in the interface.

closed 2020-05-29 by gbiggs