autowarefoundation/autoware #2182: [Feature] Velodyne VLS128 driver

alexandrx, 2019-04-01:
## Feature request

### Description
Velodyne VLS128 is not officially supported on the ROS farm, and we have no evaluation of this LiDAR on Autoware and on field tests.
This new feature proposes to enable VLS128 drivers into Autoware (while they are not on ROS farm)

[image]

### Implementation considerations
I have the working drivers in my fork https://github.com/alexandrx/velodyne_vls.git
And have already tested those with VLS128. It need some slight modifications.

### Alternatives
None

### Additional information
None

esteve, 2019-04-01:
@alexandrx I just checked your fork and it seems to be identical to upstream, did you make any changes to the driver? In any case, it'd be good to figure out what the differences are between the VLS-128 driver and the rest of the drivers and if possible, incorporate them into https://github.com/ros-drivers/velodyne

@JWhitleyAStuff do you think it'd make sense to add support for VLS-128 in the current velodyne driver?

JWhitleyAStuff, 2019-04-01:
@esteve Yes, definitely. There is not currently a ROS driver that supports the VLS-128.

gbiggs, 2019-04-02:
Please push this upstream. The upstream maintainer seems likely to accept a PR. ;)

amc-nu, 2019-04-02:
Can keep this open? I completely agree with pushing to upstream. I think @alexandrx meant to add the driver integration to Autoware (launch files, topic remapping).

alexandrx, 2019-04-02:
So how is this going to be handled? The issue title was changed but it remains closed!
And didn't we have time until May 20th (feature development freeze)? 
Between now and then we can define correctly how to proceed with the issue.

gbiggs, 2019-04-02:
> I think @alexandrx meant to add the driver integration to Autoware (launch files, topic remapping).

Sorry, I misunderstood the purpose of this feature issue. The description implied actually putting the driver into Autoware until it gets picked up by ROS, but @JWhitleyAStuff seems likely to get an update out pretty quick and syncs are frequent, so I don't think we need that.

If the purpose is editing launch files then that is obviously something that only Autoware can do. :smiley: 

> So how is this going to be handled? The issue title was changed but it remains closed!

Please don't panic. The title was changed during my morning run through updated issues. @amc-nu made his comment probably less than an hour after I closed the window.

> And didn't we have time until May 20th (feature development freeze)?

For development of features, yes. But if it's not assigned to the v1.12 milestone now we don't want to add it to the 1.12 release to prevent the scope creep that has been a regular problem in the past, except for cases that have a very good reason. If you have one of those, please make a post on Discourse about it so we can discuss it.) This process will probably chance in 1.13 but for now this is what we are going with to try and improve quality.

closed 2020-05-29 by gbiggs