autowarefoundation/autoware #2187: ERROR: cannot launch node of type [lidar_apollo_cnn_seg_detect/lidar_apollo_cnn_seg_detect]

defendergz, 2019-04-03:
## Bug report
when I click lidar_cnn_baidu_detect,  i got this:
`ERROR: cannot launch node of type [lidar_apollo_cnn_seg_detect/lidar_apollo_cnn_seg_detect]: can't locate node [lidar_apollo_cnn_seg_detect] in package [lidar_apollo_cnn_seg_detect]`

### Required information:

- Operating system and version:
  - 
   ubuntu16.04
- Autoware installation type:
  - 
from source
- Autoware version or commit hash
  - 

v 1.11
- ROS distribution and version:
  - 
kinetic
- ROS installation type:
  - 
- Package or library, if applicable:
  - 

### Description of the bug

### Steps to reproduce the bug

i have installed caffe and do `$ make distribute` correctly.
1 ./run
2 play---pause---Setup,TF,Vehicle model
3 computing tab,-----lidar detetor---lidar_cnn_baidu_detect---app---select `home/gz/apollo-cnn/cnnseg/velodyne64/deploy.prototxt` and `/home/gz/apollo-cnn/cnnseg/velodyne64/deploy.caffemodel`---OK
4 click it √
5 then i got error:

[image]

### Expected behavior

i want to get topic `/detection/lidar_detector/objects_markers` in MarkerArray in Rviz,and see sth. like this:
[image]

### Actual behavior

but i got this error `ERROR: cannot launch node of type [lidar_apollo_cnn_seg_detect/lidar_apollo_cnn_seg_detect]: can't locate node [lidar_apollo_cnn_seg_detect] in package [lidar_apollo_cnn_seg_detect]`
and cann't see anything in rviz.

who can help me? thanks a lot.

### Screenshots

### Additional information

defendergz, 2019-04-03:
@dejanpan @esteve @tushar-dadlani @gbiggs

AI-liu, 2019-04-07:
I got the same problem here.  please help!

AI-liu, 2019-04-07:
[lidar_point_pillars.launch] is neither a launch file in package [lidar_point_pillars] nor is [lidar_point_pillars] a launch file name
The traceback for the exception was written to the log file

amc-nu, 2019-04-10:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

closed 2019-04-10 by amc-nu

defendergz, 2019-04-25:
> [lidar_point_pillars.launch] is neither a launch file in package [lidar_point_pillars] nor is [lidar_point_pillars] a launch file name
> The traceback for the exception was written to the log file

any solution? i haven't yet solved it. could u help me?

gbiggs, 2019-04-26:
@defendergz You have been told what to do to get help. Please follow those instructions. Additionally, please do not open a new issue that just repeats your old one.