autowarefoundation/autoware #2193: add Lane Change in Openplanner

ChengWang2018, 2019-04-08:
## Feature request

### Description

### Implementation considerations

### Alternatives

### Additional information

ChengWang2018, 2019-04-08:
@hatem-darweesh Could you please give me some tipps how the lane change function can be added? i thought that either i have to replan the global path in case of car in the front or i generate a 5 order spline to from current lane to adjacent lane which is regarded as the reference of rollouts.

hatem-darweesh, 2019-04-11:
@ChengWang2018 
1) we need map support. (for each waypoint we can know that lane change is available). in the OpenPlanner internal map format, I have (left, right) waypoint pointers. if the pointer is not null, that means we can plan to that point. 

2) Global planner shouldn't called frequently , because it is slow and searches the whole map. solution to this, we can run global_planner at 1hz. 

3) global planner planes for final destination and all other possible paths (this is currently working) 

4) trajectory generator should generate roll outs for all global path generated. giving rollouts from different lanes higher cost (currently included) called "lane change cost)

5) trajectory evaluator (optimization function) should select one trajectory. 

6) once the vehicle changes the lane, global planner should be invoked again to calculate shortest path from this lane.

ChengWang2018, 2019-04-11:
@hatem-darweesh 
Thank you for your reply.
for the third point, i have a question. the current global planner will plan only one path if the initial position is at the same lane with the goal position, which means not all possible paths are available. in this case if there is a car in front at the same lane, the ego will follow and then stop. i think the global path should be invoked again if there is car in front to acquire all possible global paths

hatem-darweesh, 2019-04-11:
the global planning function "PlanUsingDP" in PlannerH takes a parameter "bEnableLaneChange" , if this is true, it will plan for current lane and next lanes if possible (linked)

closed 2019-04-11 by amc-nu

ChengWang2018, 2019-04-11:
@hatem-darweesh 
yes, i have enabled this "bEnableLaneChange" as true. The problem is if the initial position is at the same lane with the goal, only one global path will be generated.

ChengWang2018, 2019-04-11:
if the goal at different lane with the initial position, there will be severals paths generated. that has no problem.

DmitryRedin, 2019-04-23:
@ChengWang2018, Hello, have you solved the problem?

ChengWang2018, 2019-04-24:
> 
> 
> @ChengWang2018, Hello, have you solved the problem?

@DmitryRedin I am trying to solve. when i finish it, i will tell you

c1hong, 2020-11-13:
> > @ChengWang2018, Hello, have you solved the problem?
> 
> @DmitryRedin I am trying to solve. when i finish it, i will tell you

Hi, I have a same problem. 
Did you solved it?

JWhitleyWork, 2020-11-17:
@c1hong This issue is very old. Per our [support guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines), please ask support questions through [ROS Answers](https://answers.ros.org/) wih the `autoware` tag.