autowarefoundation/autoware #2201: After ver1.70 ndt_localizer desn't run on GPU

andoh104, 2019-04-11:
In ver.1.6.3,  ndt_localizer runs on GPU. 
But after version 1.7, ndt_localizer doesn't run on the GPU.
CUDA version is 8.0 and 9.2 .

How can it run on GPU?

This screenshot is ver.1.10.0 .
[image]

[image]

amc-nu, 2019-04-12:
Thanks for using Autoware and for your bug report.

Your report does not give us enough information to reproduce the bug. If we cannot reproduce it, we will be unable to find the cause and fix it. We ask that you please follow our support guidelines when making a bug report: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

If you think that this may not be a bug but is instead a problem in your use of Autoware, then consider closing this bug report and opening a topic on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) requesting help. This may get you past your problem faster. Link to your new topic from this issue so we know where to continue the discussion.

andoh104, 2019-04-12:
Thank you for your reply.

Information to repruduce:
- Ubuntu 16.04
- ROS kinetic
- CUDA 8.0 and 9.2 (CUDA 10.1 could not be built)
- Autoware version 1.10.0, 1.7.0 (1.6.3 is OK)

I can provide the additional information if you need.

Please provide information that this is restriction of Autoware or not.
Because I can’t judge whether this is bug or not.

amc-nu, 2019-04-12:
Please also upload your `ndt_gpu` and `lidar_localizer` build logs. 
Located in the `log/latest` directory.

andoh104, 2019-04-12:
log/latest/ndt_gpu
- [command.log](https://github.com/autowarefoundation/autoware/files/3073598/command.log)
- [stderr.log](https://github.com/autowarefoundation/autoware/files/3073599/stderr.log)
- [stdout.log](https://github.com/autowarefoundation/autoware/files/3073600/stdout.log)
- [stdout_stderr.log](https://github.com/autowarefoundation/autoware/files/3073601/stdout_stderr.log)
- [streams.log](https://github.com/autowarefoundation/autoware/files/3073602/streams.log)

There is no lidar_localizer directry.
[image]

amc-nu, 2019-04-12:
Thanks. I confirmed ndt is being compiled with GPU support.
Can you please tell me
1. How are you compiling?
2. Can you please also show the output of the Autoware terminal after you launch ndt from runtime manager?

Thanks

andoh104, 2019-04-12:
This is screenshot after starting Localization. 
The Localization is matching. I confirmed by RViz.
[image]

THanks

andoh104, 2019-04-14:
I couldn't test in Autoware 1.11.0, because of this issue #2202 .

andoh104, 2019-04-15:
Information to repruduce
- Ubuntu 16.04
- ROS kinetic
- CUDA 8.0 and 9.2 (CUDA 10.1 could not be built)
- Autoware version 1.10.0, 1.7.0 (1.6.3 is OK)

Please provide the information that this is restriction of Autoware or not.
(I can’t jduge this is bug or not)

amc-nu, 2019-04-15:
@andoh104 This is not a bug.

The default method is CPU.
You'll need to set to GPU the NDT method.
https://github.com/autowarefoundation/autoware/blob/master/ros/src/computing/perception/localization/packages/lidar_localizer/launch/ndt_matching.launch#L4
I recommend not to use "quick start" launch files.

closed 2019-04-15 by amc-nu

andoh104, 2019-04-15:
@amc-nu 
Thank you. I understand default setting.

Could you show me how to set to GPU,  Is this OK ?
` <arg name="method_type" default="3" /> `

I tried this, but it seems not to run on GPU.

gbiggs, 2019-04-16:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

andoh104, 2019-05-05:
@gbiggs 
Thank you.
I found this topic on the Autoware Discourse category .
https://discourse.ros.org/t/ndt-mapping-method-types-and-parameters/8482/3

I think that since  NDT localizer on GPU has not been supported and not tested since  v1.6 , so this issue is not a bug.
Is this my understanding correct ?