autowarefoundation/autoware #2202: Invalid roslaunch XML syntax: [Errno 2] No such file or directory

andoh104, 2019-04-14:
## Bug report

### Required information:

- Operating system and version: Ubuntu 16.04
- Autoware installation type: From source
- Autoware version: 1.11.0 and master and develop (1.10.0 is OK)
- ROS distribution and version: kinetic
- ROS installation type: from binaries
- Package or library, if applicable: CUDA 9.2

### Description of the bug

After starting Localization from Runtime Manager, this error occurs.

    ['roslaunch', '/home/xxx/Autoware/docs/quick_start/my_localization.launch']
    pid=19626
    sched policy=OTHER prio=0
    while processing /home/xxx/Autoware/ros/install/runtime_manager/share/runtime_manager/scripts/setup_tf.launch:
    Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/xxx/Autoware/ros/install/runtime_manager/share/runtime_manager/scripts/setup_tf.launch'
    The traceback for the exception was written to the log file

### Steps to reproduce the bug
1. ./run
1. Simulation -> Play
1. QuickStart -> Map
1. QuickStart ->  Localization
1. See above error displayed

### Expected behavior
Launch Localization

### Actual behavior

Can'' launch Localization
See above error.

### Screenshots

### Additional information

In my environment "setup_tf.launch" don't exist in 
"ros/install/runtime_manager/share/runtime_manager/scripts/"
Actually it is here:
"ros/install/runtime_manager/share/runtime_manager/launch_files/setup_tf.launch"

sgermanserrano, 2019-04-15:
@andoh104 this is currently being addressed as part of #2176

sgermanserrano, 2019-04-18:
@andoh104 the bug as been solved as part of #2206. Can you try using the updated quick_start files?

closed 2019-04-18 by sgermanserrano

andoh104, 2019-05-05:
@sgermanserrano 
I tested the master repository and it was OK.
Thanks,