autowarefoundation/autoware #2221: roslaunch autoware_camera_lidar_calibrator

EricWu123, 2019-04-19:
## Bug report
... logging to /home/wyq/.ros/log/0d459c74-623b-11e9-bd8f-0090f5fb14b2/roslaunch-wyq-CW35S-9307.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://wyq-CW35S:45999/

SUMMARY
========

PARAMETERS
 * /autoware_camera_lidar_calibration_node/camera_info_src: /camera_info
 * /autoware_camera_lidar_calibration_node/image_src: /image_rectified
 * /calibration_publisher/calibration_file: /home/wyq/2019041...
 * /calibration_publisher/camera_frame: camera
 * /calibration_publisher/image_topic_src: /image
 * /calibration_publisher/publish_camera_info: True
 * /calibration_publisher/publish_extrinsic_mat: False
 * /calibration_publisher/register_lidar2camera_tf: False
 * /calibration_publisher/target_frame: velodyne
 * /rectifier/camera_info_src: /camera_info
 * /rectifier/image_src: /image
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    autoware_camera_lidar_calibration_node (autoware_camera_lidar_calibrator/autoware_camera_lidar_calibration_node)
    calibration_publisher (calibration_publisher/calibration_publisher)
    image_view2 (image_view2/image_view2)
    rectifier (image_processor/image_rectifier)

ROS_MASTER_URI=http://localhost:11311

process[calibration_publisher-1]: started with pid [9324]
process[rectifier-2]: started with pid [9325]
process[image_view2-3]: started with pid [9340]
[ INFO] [1555636984.914030462]: [image_rectifier] image_src: /image
[ INFO] [1555636984.914123850]: [image_rectifier] camera_info_src: /camera_info
[ INFO] [1555636984.914140007]: [image_rectifier] Subscribing to... /image
ERROR: cannot launch node of type [autoware_camera_lidar_calibrator/autoware_camera_lidar_calibration_node]: can't locate node [autoware_camera_lidar_calibration_node] in package [autoware_camera_lidar_calibrator]
[ INFO] [1555636984.917238717]: [image_rectifier] Subscribing to... /camera_info
[ INFO] [1555636984.921328329]: [image_rectifier] Publishing Rectified image in /image_rectified
[ INFO] [1555636984.921387237]: [image_rectifier] Ready. Waiting for data...
### Required information:

When I run the code
```
roslaunch autoware_camera_lidar_calibrator camera_lidar_calibration.launch intrinsics_file:=/home/wyq/20190417_1624_autoware_camera_calibration.yaml image_src:=/image 
```
ERROR: cannot launch node of type [autoware_camera_lidar_calibrator/autoware_camera_lidar_calibration_node]: can't locate node [autoware_camera_lidar_calibration_node] in package [autoware_camera_lidar_calibrator]
 happened.
It seems that the problem of environment variables. I set environment variables by source install/setup.bash, but I don't know what wrong it is.

mitsudome-r, 2019-04-19:
It seems like install option in CMakeLists.txt is wrong.
Try adding these lines at the end of ~/Autoware/ros/src/sensing/fusion/packages/autoware_camera_lidar_calibrator/CMakeLists.txt. 
and recompile. 
```
install(TARGETS autoware_camera_lidar_calibration_node
        ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
        LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
        RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)
```

EricWu123, 2019-04-19:
thank u~, but i just copy the executable file ' autoware_camera_lidar_calibration_node ' in the folder ' Autoware/ros/build/autoware_camera_lidar_calibrator/devel/lib/autoware_camera_lidar_calibrator ', to the folder 'Autoware/ros/install/autoware_camera_lidar_calibrator/lib/autoware_camera_lidar_calibrator', and solve this problem.

amc-nu, 2019-04-19:
already fixed on #2195

closed 2019-04-19 by amc-nu