autowarefoundation/autoware #2222: Migrate project to autowarefoundation in Gitlab

esteve, 2019-04-19:
Mirroring https://gitlab.com/AutowareAuto/AutowareAuto/issues/79

The Autoware foundation now qualifies as a NPO for Gitlab's OSS program, therefore we should be moving this project to the appropriate organization.

- [ ] Move projects to https://gitlab.com/autowarefoundation
- [ ] Set up CI. Gitlab's CI infrastructure does not support the fork + pull model that GitHub users are used to (see https://gitlab.com/gitlab-org/gitlab-ce/issues/21991). We can integrate Gitlab with our CI by deploying their runner software  on our own infrastructure and with the Webhooks API https://docs.gitlab.com/ee/user/project/integrations/webhooks.html#merge-request-events

/cc @amc-nu @gbiggs @sgermanserrano

gbiggs, 2019-04-24:
Can GitLab import issues as well?

esteve, 2019-04-24:
@gbiggs yes, but for mapping GitHub users to Gitlab's properly, we need to use the GitHub integration, detailed in https://docs.gitlab.com/ee/user/project/import/github.html#using-the-github-integration

kennedywai, 2019-05-31:
Has Github repo already been abandoned?

gbiggs, 2019-05-31:
Not yet but we plan to finalise the move in the next few weeks.

kennedywai, 2019-05-31:
@gbiggs Its been 1.5 months since the last someone asked an issue here. I thought the gitbub was closed.

gbiggs, 2019-06-03:
There have been plenty of issues and PRs. However most issues have been closed because we have moved support to ROS Answers. See the support guide for more information.

kennedywai, 2019-06-03:
@gbiggs  Cool! I didnt realize it!

kennedywai, 2019-06-03:
For those we are looking for related autoware issues, please check out the link below: 
[](https://answers.ros.org/questions/scope:all/sort:activity-desc/page:1/query:autoware/)

closed 2020-05-29 by gbiggs