autowarefoundation/autoware #2224: Size of detected_objects_output  is always 1 in IMM_UKF_PDA TRACK

saii95, 2019-04-21:
After passing the transformed_input to the tracker. size of transformed_input  is 18 varying time to time. but the size of the detected_objects_output is always 1. After passing this detected_objects_output to the visualizers node to visualize in RVIZ.  suppose if there are 'n' number of clusters.  it is able to draw a convex hull to only  one the cluster. Among that 'n' number of clusters.

gbiggs, 2019-04-21:
Thanks for using Autoware and for your bug report.

Your report does not give us enough information to reproduce the bug. If we cannot reproduce it, we will be unable to find the cause and fix it. We ask that you please follow our support guidelines when making a bug report: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

If you think that this may not be a bug but is instead a problem in your use of Autoware, then consider closing this bug report and opening a topic on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) requesting help. This may get you past your problem faster. Link to your new topic from this issue so we know where to continue the discussion.

saii95, 2019-04-22:
I did the same but i didnt get any reponse so i am posting here. Sorry for the inconvenience.

Here is the callback which i am subscrbing cloudcluters from lidar_euclidean_cluster_detect node. And giving that clusters for tracking.

void ImmUkfPda::callback(const autoware_msgs::DetectedObjectArray& input)
{
  input_header_ = input.header;

  if(use_vectormap_)
  {
    checkVectormapSubscription();
  }

  bool success = updateNecessaryTransform();
  if (!success)
  {
    ROS_INFO("Could not find coordiante transformation");
    return;
  }

  autoware_msgs::DetectedObjectArray transformed_input;
  autoware_msgs::DetectedObjectArray detected_objects_output;
  transformPoseToGlobal(input, transformed_input);
  tracker(transformed_input, detected_objects_output);
  transformPoseToLocal(detected_objects_output);

  pub_object_array_.publish(detected_objects_output);

  if (is_benchmark_)
  {
    dumpResultText(detected_objects_output);
  }
}

I mean to say that the of  size of "detected_objects_output" is always  "1"   This is used for visualise the tracked objects in [RVIZ.]

Please find the below screen shot for output of IMM_UKF_PDA_TRACK. Although there are multiple number of clusters around the ego vechile. it is tracking only one object at a time. i mean suppose if there is object in right and left . It is tracking only  right object after few seconds it is tracking left object but not the right side object.   

[image]

k0suke-murakami, 2019-04-22:
Hi,
As Geoff said, we need more information to reproduce your problem.

Please fill the below form.

## Bug report

### Required information:

- Operating system and version:
  - 
- Autoware installation type:
  - 
- Autoware version or commit hash
  - 
- ROS distribution and version:
  - 
- ROS installation type:
  - 
- Package or library, if applicable:
  - 

### Description of the bug

### Steps to reproduce the bug

### Expected behavior

### Actual behavior

### Screenshots

### Additional information

saii95, 2019-04-22:
## Bug report

### Required information:

- Operating system and version:
  - Ubuntu 16.04-->
- Autoware installation type:
  - Directly cloned the workspace and taken the tracker node.-->
- Autoware version or commit hash
  - VER.1.11.0
- ROS distribution and version:
  - 
- ROS installation type:
  - 
- Package or library, if applicable:
  - 

### Description of the bug

### Steps to reproduce the bug

### Expected behavior

### Actual behavior

### Screenshots

saii95, 2019-04-22:
## Bug report

### Required information:

- Operating system and version:
  - Ubuntu 16.04
- Autoware installation type:
  - Directly downloaded workspace and taken the tracker node 
- Autoware version or commit hash
  - Ver 1.11.0
- ROS distribution and version:
  - Ros kinetic
- ROS installation type:
  -From source
- Package or library, if applicable:
  -  N/A 

### Description of the bug

- please find the below attached piece code for better understanding

### Expected behavior

### Actual behavior

### Screenshots

### Additional information

gbiggs, 2019-04-22:
> I did the same but i didnt get any reponse so i am posting here. Sorry for the inconvenience.

Please don't assume that not receiving an instant reply means you will not get a response. As it happens, your new topic at Discourse has only just been released from moderation. I recommend you close this issue and discuss your problem at Discourse, and if a bug is confirmed then we can re-open this issue with the new information. I think you will get a faster response that way. More people watch Discourse than watch the issue tracker.

saii95, 2019-04-23:
Sure

closed 2019-04-23 by saii95