autowarefoundation/autoware #2226: ERROR: cannot launch node of type [lidar_apollo_cnn_seg_detect/lidar_apollo_cnn_seg_detect]

defendergz, 2019-04-25:
when I click lidar_cnn_baidu_detect, i got this:
ERROR: cannot launch node of type [lidar_apollo_cnn_seg_detect/lidar_apollo_cnn_seg_detect]: can't locate node [lidar_apollo_cnn_seg_detect] in package [lidar_apollo_cnn_seg_detect]

Operating system and version:
ubuntu16.04

i have installed caffe and do $ make distribute correctly.
1 ./run
2 play---pause---Setup,TF,Vehicle model
3 computing tab,-----lidar detetor---lidar_cnn_baidu_detect---app---select home/gz/apollo-cnn/cnnseg/velodyne64/deploy.prototxt and /home/gz/apollo-cnn/cnnseg/velodyne64/deploy.caffemodel---OK
4 click it √
5 then i got error:
[image]

Expected behavior
i want to get topic /detection/lidar_detector/objects_markers in MarkerArray in Rviz,and see sth. like this:
[image]

Actual behavior
but i got this error ERROR: cannot launch node of type [lidar_apollo_cnn_seg_detect/lidar_apollo_cnn_seg_detect]: can't locate node [lidar_apollo_cnn_seg_detect] in package [lidar_apollo_cnn_seg_detect]
and cann't see anything in rviz.

who can help me? thanks a lot.

defendergz, 2019-04-25:
@dejanpan @esteve @tushar-dadlani

amc-nu, 2019-04-25:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

closed 2019-04-25 by amc-nu