autowarefoundation/autoware #2229: Time of of Localization is not shown in Runtime Manager/Quickstart

andoh104, 2019-04-25:
## Bug report

### Required information:

- Operating system and version:Ubuntu 16.04
- Autoware installation type:From source
- Autoware version or commit hash
  - 1.7.0, 1.9.0, 1.9.1. 10.0)
  - 1.6.3, 1.80 are OK
  -  Version 1.11 and later can not test because of another problem.

- ROS distribution and version:kinetic
- ROS installation type:from binaries
- Package or library, if applicable:N/A

### Description of the bug
Time of of Localization is not shown in Runtime Manager/Quickstart
This issue occurs in version 1.7.0 and later.

### Steps to reproduce the bug
1. ./run
1. Simulation -> Play
1. Quick Start -> Map
1. Simulation -> Pause
1. Quick Start ->  Localization
1. Simulation -> Pause

### Expected behavior
Time of of Localization is displayed in Runtime Manager /Quickstart
[image]

### Actual behavior
Time of of Localization is not shown in Runtime Manager /Quickstart
[image]

### Screenshots

### Additional information

I think these are wrong:
- https://github.com/autowarefoundation/autoware/commit/86f74272b0d933abb4b5d4e0a28802592843f4f1#diff-994bc52df555b725fcb3d7b458fcf90eL92
`from` autoware_msgs.msg import NDTStat

- https://github.com/autowarefoundation/autoware/blob/master/ros/src/util/packages/runtime_manager/scripts/qs.yaml#L3
`/ndt_stat.NDTStat.exe_time` :

andoh104, 2019-04-29:
I made fixed branch from master.
https://github.com/andoh104/Autoware/tree/fix/runtime_manager_not_show_time
https://github.com/andoh104/Autoware/commit/0887af0f59408b70eaf8994b126590ddd1a809f3

gbiggs, 2019-04-29:
If you have a fix, then please create a PR for it.

andoh104, 2019-05-05:
@gbiggs 
I made PR to master.
#2235

andoh104, 2019-05-18:
PR was merged, so this issue is closed.

closed 2019-05-18 by andoh104