autowarefoundation/autoware #2232: Build error

jinfagang, 2019-04-28:
```
ment&)':
nmea2kml.cpp:(.text+0x183): undefined reference to `TiXmlDeclaration::TiXmlDeclaration(char const*, char const*, char const*)'
nmea2kml.cpp:(.text+0x19f): undefined reference to `TiXmlElement::TiXmlElement(char const*)'
nmea2kml.cpp:(.text+0x1b5): undefined reference to `TiXmlElement::SetAttribute(char const*, char const*)'
nmea2kml.cpp:(.text+0x1cb): undefined reference to `TiXmlElement::SetAttribute(char const*, char const*)'
nmea2kml.cpp:(.text+0x1e1): undefined reference to `TiXmlElement::SetAttribute(char const*, char const*)'
nmea2kml.cpp:(.text+0x1f7): undefined reference to `TiXmlElement::SetAttribute(char const*, char const*)'
nmea2kml.cpp:(.text+0x213): undefined reference to `TiXmlElement::TiXmlElement(char const*)'
nmea2kml.cpp:(.text+0x21e): undefined reference to `TiXmlNode::LinkEndChild(TiXmlNode*)'
nmea2kml.cpp:(.text+0x23a): undefined reference to `TiXmlElement::TiXmlElement(char const*)'
nmea2kml.cpp:(.text+0x246): undefined reference to `TiXmlNode::LinkEndChild(TiXmlNode*)'
nmea2kml.cpp:(.text+0x260): undefined reference to `TiXmlNode::TiXmlNode(TiXmlNode::NodeType)'
nmea2kml.cpp:(.text+0x267): undefined reference to `vtable for TiXmlText'
nmea2kml.cpp:(.text+0x29b): undefined reference to `TiXmlNode::LinkEndChild(TiXmlNode*)'
nmea2kml.cpp:(.text+0x2b7): undefined reference to `TiXmlElement::TiXmlElement(char const*)'
nmea2kml.cpp:(.text+0x2c3): undefined reference to `TiXmlNode::LinkEndChild(TiXmlNode*)'
nmea2kml.cpp:(.text+0x2dd): undefined reference to `TiXmlNode::TiXmlNode(TiXmlNode::NodeType)'
nmea2kml.cpp:(.text+0x2e4): undefined reference to `vtable for TiXmlText'
nmea2kml.cpp:(.text+0x318): undefined reference to `TiXmlNode::LinkEndChild(TiXmlNode*)'
nmea2kml.cpp:(.text+0x323): undefined reference to `TiXmlNode::LinkEndChild(TiXmlNode*)'
nmea2kml.cpp:(.text+0x356): undefined reference to `TiXmlNode::~TiXmlNode()'
CMakeFiles/nmea2kml.dir/nodes/nmea2kml/nmea2kml.cpp.o: In function `CreateStyleWithColor(TiXmlElement*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int, int, int)':
nmea2kml.cpp:(.text+0x663): undefined reference to `TiXmlElement::TiXmlElement(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
nmea2kml.cpp:(.text+0x66e): undefined reference to `TiXmlNode::LinkEndChild(TiXmlNode*)'
nmea2kml.cpp:(.text+0x6a9): undefined reference to `TiXmlElement::TiXmlElement(char const*)'
nmea2kml.cpp:(.text+0x6b4): undefined reference to `TiXmlNode::LinkEndChild(TiXmlNode*)'
nmea2kml.cpp:(.text+0x6df): undefined reference to `TiXmlNode::TiXmlNode(TiXmlNode::NodeType)'
nmea2kml.cpp:(.text+0x6e6): undefined reference to `vtable for TiXmlText'
nmea2kml.cpp:(.text+0x709): undefined reference to `TiXmlNode::LinkEndChild(TiXmlNode*)'
nmea2kml.cpp:(.text+0x88a): undefined reference to `TiXmlNode::~TiXmlNode()'
CMakeFiles/nmea2kml.dir/nodes/nmea2kml/nmea2kml.cpp.o: In function `InsertNewStyle(TiXmlElement*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int, int)':
nmea2kml.cpp:(.text+0x91a): undefined reference to `TiXmlElement::TiXmlElement(char const*)'
nmea2kml.cpp:(.text+0x925): undefined reference to `TiXmlNode::LinkEndChild(TiXmlNode*)'
nmea2kml.cpp:(.text+0x968): undefined reference to `TiXmlElement::SetAttribute(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<cha
```

While already has tinyxml installed. any suggestions?

gbiggs, 2019-04-29:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

closed 2019-04-29 by gbiggs

DavidNY123, 2022-03-22:
Hello, I solve this problem by a simple method.
autoware.ai/ros/src/util/packages/autoware_bag_tools/nodes/nmea2kml/nmea2kml.cpp
I changed the order of header files，It looks like：
#include <ros/ros.h>
#include <rosbag/bag.h>
#include <rosbag/view.h>
#include <tinyxml.h>   

#include <string>
#include <vector>
#include <nmea_msgs/Sentence.h>
#include "ReadNMEASentence.h"
#include <iostream>
#include <fstream>
#include <sstream>
#include <ostream>
#include <boost/foreach.hpp>
#include <boost/filesystem.hpp>

#include "dirent.h"