autowarefoundation/autoware #2236: Docker quick start file not found

jinfagang, 2019-05-05:
Pulled newest autoware docker, runing with error:

[image]

localization can not be complete.

jinfagang, 2019-05-05:
```
[Errno 2] No such file or directory: u'/home/autoware/Autoware/ros/install/runtime_manager/share/runtime_manager/scripts/setup_tf.launch'

```

YangLiu14, 2019-05-05:
Hi, I have the same issue, and it turns out that the path is wrong in docker. See my post here:
https://discourse.ros.org/t/autoware-1-11-0-quick-start-demo-failed-to-load-the-map/9019

I also tried source build in my ubuntu 16 machine, and the path is actually correct for source build, I wonder why it is different in docker?

jinfagang, 2019-05-05:
@YangLiu14  What do you mean path correct in ubuntu 16.04? Dokcer path has anything to do with host machine?

YangLiu14, 2019-05-05:
Don't worry about the thing I said about Ubuntu 16, I just tried to source build Autoware there.

I just meant the in the `my_localiztion.launch` file the path for finding the `setup_tf.launch` is ".../runtime_manager/**script**/setup_tf.launch".   This works fine, when I source build the Autoware from master-branch in my host.       But, when I run things in docker, the path is incorrect now because `setup_tf.launch` is located in ".../runtime_manager/**launch_files**/setup_tf.launch" inside the docker.

So I guess, it might simply because they didn't update the path in launch file inside the docker.

jinfagang, 2019-05-06:
@YangLiu14 Right. anyway, the newest docker image even the default rviz setting were not right (it does not using base_link or world but map). Why they build docker image without any test........

gbiggs, 2019-05-07:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

We are aware of the several issues with paths in the 1.11 release. They are being fixed in the repository. Contributions of bug fixes and testing are always welcome.

jinfagang, 2019-05-07:
it is a bug, pls confirm

gbiggs, 2019-05-07:
Yes, it is a bug, and it has [been fixed](https://github.com/autowarefoundation/autoware/pull/2206).

gbiggs, 2019-05-07:
Please ask your question at [ROS Answers](https://answers.ros.org/questions/ask/?tags=autoware).

kfunaoka, 2019-05-16:
It seems this issue can be closed since (1) the bug is fixed, and (2) ROS Answers is introduced.

closed 2019-05-16 by kfunaoka