autowarefoundation/autoware #2250: How to train point pillars

jinfagang, 2019-05-14:
Since point pillars have been released on SECOND repo. is that possiable share some scripts to export model trained on pytorch to TensorRT support onnx?

gbiggs, 2019-05-14:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

closed 2019-05-14 by gbiggs

ryontang, 2021-01-13:
> Since point pillars have been released on SECOND repo. is that possiable share some scripts to export model trained on pytorch to TensorRT support onnx?

Hi, Have you trained Pointpillars and located its ONNX file in Autoware, if you did, it is helpful to me for you giving some references