autowarefoundation/autoware #2260: Don't install library gstreamer0.10-plugins-good in ubuntu 18.04

duclinhfetel, 2019-05-15:
## Bug report

### Required information:

- Operating system and version:
  -  OS: Ubuntu 18.04
- Autoware installation type:
  - From source: https://github.com/autowarefoundation/autoware/wiki/Source-Build
- Autoware version or commit hash
  - git clone https://github.com/CPFL/Autoware.git --recurse-submodules
- ROS distribution and version:
  - rosdistro: melodic
  - rosversion: 1.14.3

### Description of the bug

  - Use command: rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
  - Don't install **gstreamer0.10-plugins-good**

### Screenshots

[image]

gbiggs, 2019-05-16:
Thanks for using Autoware and for your bug report.

I think that this is most likely to be a problem on your computer or possibly with ROS, rather than a bug in Autoware. In such cases we prefer to first identify the cause of the problem at ROS Answers and then file a bug report if it is actually confirmed to be a bug.

Please ask your question at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide. If it is confirmed to be a bug in Autoware, we will re-open this issue.

closed 2019-05-16 by gbiggs

Jameswz, 2019-12-15:
Edit //etc/apt/sources.list and add below repos . your issue will be fixed. 
deb http://security.ubuntu.com/ubuntu xenial-security main universe
deb http://cz.archive.ubuntu.com/ubuntu xenial main universe