autowarefoundation/autoware #2300: Autoware Docker Setup sh instead bash

pamarinp, 2019-06-06:
## Bug report
Wiki error in section Case 1: Using Docker Hub "As Is" and following
### Required information:

- Operating system and version: Ubuntu 16.04
  - 
- Autoware installation type: Docker + nvidia plugin
  - 
- Autoware version or commit hash
  - 
- ROS distribution and version: Kinetic
  - 
- ROS installation type: Full desktop
  - 
- Package or library, if applicable:
  - 

### Description of the bug

In the tutorial, following the steps, it is said: 
sudo sh run.sh
When should be said sudo bash run.sh (because is bash)

### Steps to reproduce the bug

$ head run.sh 

```
#!/bin/bash

# Default settings
CUDA="on"
IMAGE_NAME="autoware/autoware"
TAG_PREFIX="latest"
ROS_DISTRO="kinetic"
BASE_ONLY="false"
PRE_RELEASE="off"
AUTOWARE_GROUP_ID=15214
```

### Expected behavior

### Actual behavior

### Screenshots

### Additional information

gbiggs, 2019-06-06:
Please post this as a bug against the [GitLab repository](https://gitlab.com/autowarefoundation/autoware.ai/autoware/). This repository will be changed to a mirror soon.

gbiggs, 2019-06-06:
Please make a pull request against the wiki repository to fix this bug.

https://github.com/autowarefoundation/autoware-wiki

closed 2019-06-06 by gbiggs

italojs, 2019-06-15:
I made a pull request to fix it but I used a better approach 
Update the documentation will deprecate the currently videos/articles in the community, so it's better to update the code.
PR: #2305