autowarefoundation/autoware #2301: default rviz file wants wants topic op_trajectory_generator to have message type visualization_msgs/MarkerArray instead of autoware_msgs/LaneArray

hroachewilson, 2019-06-10:
## Bug report
Rviz isn't compiled with autoware_msgs/LaneArray message, preventing visualisation of local_trajectories topic published by op_trajectory_generator

### Required information:

- Operating system and version:
Ubuntu 16.04
- Autoware installation type:
Docker container built from source
- Autoware version or commit hash
[2b8fb78](https://github.com/hroachewilson/Autoware/commit/2b8fb78ddd7f831a77aa8addccb82cb5ebc5cbd2) (Feb 3, 2019).
- ROS distribution and version:
`autoware@rover:~/Autoware/ros/src$ rosversion --distro
kinetic`
`autoware@rover:~/Autoware/ros/src$ rosversion roscpp
1.12.14`
`autoware@rover:~/Autoware/ros/src$ rosversion rviz
1.12.17`
- ROS installation type:
Generic docker file
- Package or library, if applicable:
  - rviz
  - op_trajectory_generator

### Description of the bug
When running the default moriyama bag and quick_launch files, the following error occurs on attempting to visualise Local Waypoints using `ros/src/.config/rviz/default.rviz`:

`
[ERROR] [1560131609.026609710, 1427157758.451910501]: Client [/rviz_1560131602457053824] wants topic /behavior_state to have datatype/md5sum [visualization_msgs/Marker/4048c9de2a16f4ae8e0538085ebf1b97], but our version has [visualization_msgs/MarkerArray/d155b9ce5188fbaf89745847fd5882d7]. Dropping connection.
`

Attempts to fix the error by substituting `rviz/MarkerArray` with `autoware_msgs/LaneArray` in default.rviz fail with:

`
[ERROR] [1560131608.179664397]: PluginlibFactory: The plugin for class 'autoware_msgs/LaneArray' failed to load.  Error: According to the loaded plugin descriptions the class autoware_msgs/LaneArray with base class type rviz::Display does not exist. Declared types are  grid_map_rviz_plugin/GridMap jsk_rviz_plugin/AmbientSound jsk_rviz_plugin/BoundingBox jsk_rviz_plugin/BoundingBoxArray jsk_rviz_plugin/CameraInfo jsk_rviz_plugin/Diagnostics jsk_rviz_plugin/Footstep jsk_rviz_plugin/NormalDisplay jsk_rviz_plugin/OverlayCamera jsk_rviz_plugin/OverlayDiagnostic jsk_rviz_plugin/OverlayImage jsk_rviz_plugin/OverlayMenu jsk_rviz_plugin/OverlayText jsk_rviz_plugin/PeoplePositionMeasurementArray jsk_rviz_plugin/Pictogram jsk_rviz_plugin/PictogramArray jsk_rviz_plugin/PieChart jsk_rviz_plugin/Plotter2D jsk_rviz_plugin/PolygonArray jsk_rviz_plugin/PoseArray jsk_rviz_plugin/QuietInteractiveMarker jsk_rviz_plugin/SegmentArray jsk_rviz_plugin/SimpleOccupancyGridArray jsk_rviz_plugin/TFTrajectory jsk_rviz_plugin/TargetVisualizer jsk_rviz_plugin/TorusArray jsk_rviz_plugin/TwistStamped jsk_rviz_plugin/VideoCapture rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_imu_plugin/Imu rviz_plugin_tutorials/Imu
`

### Steps to reproduce the bug

1. Run the my_map, my_sensing, my_localisation, my_detection quick_start launch files after loading the default moriyama bag. 
2. Add

`
  
  <include file="$(find lidar_kf_contour_track)/launch/lidar_kf_contour_track.launch">
    <arg name="tracking_type" value="1" />
  </include>
`

to my_mission_planning launch file so that it supports the local planner requirements

3. Run all OpenPlanner nodes using Runtime Manager (op_common_params, op_trajectory_planner, op_motion_predictor, op_trajectory_evaluator, op_behavior_selector)

4. See the error displayed.

### Expected behavior
Would expect either:
rviz to be compiled to support autoware_msgs/LaneArray so that the .rviz file recognises it, or local_trajectories to publish rviz/MarkerArray for visualisation.

### Actual behavior
```
[ERROR] [1560131609.127641052, 1427157758.451910501]: Client [/rviz_1560131602457053824] wants topic /local_trajectories to have datatype/md5sum [visualization_msgs/MarkerArray/d155b9ce5188fbaf89745847fd5882d7], but our version has [autoware_msgs/LaneArray/ca01c32675aeaa32240193d4d3f7e07a]. Dropping connection.
 ```

saitejavn, 2020-03-05:
Any update on this?

SenRamakri, 2020-05-28:
I ran into the same problem and below is what resolved it.
The problem is caused by Rviz config file having the wrong configuration for local_trajectories.
With the latest version of Autoware, topic for Rviz is named a /local_trajectories_gen_rviz (see here: )https://gitlab.com/autowarefoundation/autoware.ai/core_planning/-/blob/master/op_local_planner/nodes/op_trajectory_generator/op_trajectory_generator_core.cpp#L41).
So open your rviz config file and find the entry for local_trajectories and change it to local_trajectories_gen_rviz. This should resolve the above error.

closed 2020-05-29 by gbiggs