autowarefoundation/autoware #2302: fixframe [map] move alone with base_link

hamkuchan, 2019-06-12:
## Bug report

### Required information:

- Operating system and version:Ubuntu 16.04
  - 
- Autoware installation type:From source
  - 
- Autoware version or commit hash: Autoware Ver. 1.11.0
  - 
- ROS distribution and version: Kinetic
  - 
- ROS installation type: From source
  - 
- Package or library, if applicable:
  - 

### Description of the bug

when the base_link, the fixed frame [map] move alone with frame [base_link]

### Steps to reproduce the bug

### Expected behavior

the frame[map] should not move alone with frame [base_link].

### Actual behavior

the frame [map] move alone with frame [base_link].

### Screenshots

### Additional information

YamatoAndo, 2019-06-12:
@hamkuchan Hi.
Please share a screenshot or video of rviz.

gbiggs, 2019-06-13:
Thanks for using Autoware and for your bug report.

Your report does not give us enough information to reproduce the bug. If we cannot reproduce it, we will be unable to find the cause and fix it. We ask that you please follow our support guidelines when making a bug report: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide there and in the issue template.

If you think that this may not be a bug but is instead a problem in your use of Autoware, then consider closing this bug report and asking a question at [ROS Answers](https://answers.ros.org/questions/ask/?tags=autoware) requesting help. This may get you past your problem faster. Link to your question from this issue so we know where to continue the discussion.

hamkuchan, 2019-06-14:
> @hamkuchan Hi.
> Please share a screenshot or video of rviz.

thanks for reply, I added a tf from [map] to [odom], then it works.

hamkuchan, 2019-06-14:
> Thanks for using Autoware and for your bug report.
>
> Your report does not give us enough information to reproduce the bug. If we cannot reproduce it, we will be unable to find the cause and fix it. [...]

thanks for the reply, I'll close it.

closed 2019-06-14 by hamkuchan