autowarefoundation/autoware #2306: AUTOWARE_COMPILE_WITH_CUDA=1 fails to build

ROS-lover, 2019-06-17:
### Required information:

- Operating system and version:
  - Ubuntu 16.04 - CUDA 10.1 - tensorrt 5.1.5.0-1+cuda10.1
  - Autoware commit 42aba1468490e2ed218f9295ab3e7490db864e54
  - git tag: 1.12.0-alpha.1

- Autoware installation type:
  - from source

### Description of the bug
'$AUTOWARE_COMPILE_WITH_CUDA=1 ./colcon_release' fails to build although without this flag building finishes with a success but it says 'CUDA was not found' and for instance ndt_gpu cannot be built. (terminal's output is attached)

ssdcafe was installed sucessfully as well as CUDA. (tested with samples-cuda10.1)

Thanks in advance!

Cheers,
[autoware_noCuda.txt](https://github.com/autowarefoundation/autoware/files/3294941/autoware_noCuda.txt)

## Terminal output
ocp@ocp:~/Autoware/ros$ AUTOWARE_COMPILE_WITH_CUDA=1 ./colcon_release
Starting >>> autoware_build_flags
Starting >>> autoware_msgs
Starting >>> autoware_config_msgs
Starting >>> tablet_socket_msgs
Starting >>> autoware_system_msgs
Starting >>> autoware_can_msgs
Starting >>> gnss
Starting >>> rslidar_msgs
Starting >>> sick_ldmrs_description
Starting >>> sick_ldmrs_msgs
Starting >>> ds4_msgs
Starting >>> custom_msgs
Starting >>> ds4_driver
Starting >>> gazebo_camera_description
Starting >>> gazebo_imu_description
Starting >>> ndt_cpu
Finished <<< gazebo_camera_description [7.53s]                                                                                                                                                                  
Starting >>> ndt_tku
Finished <<< gazebo_imu_description [7.59s]                                                                                                                                      
Finished <<< autoware_build_flags [7.61s]
Starting >>> vector_map_msgs
Starting >>> pcl_omp_registration
Finished <<< sick_ldmrs_description [7.70s]                                                                                                                                                                 
Starting >>> fastvirtualscan
Finished <<< ds4_driver [7.75s]
Starting >>> kitti_player
Finished <<< rslidar_msgs [8.13s]                                                                                                                                                                           
Starting >>> rslidar_driver
Finished <<< gnss [8.16s]
Starting >>> ndt_gpu
Finished <<< tablet_socket_msgs [8.35s]                                                                                                                                                                      
Starting >>> rosinterface
Finished <<< autoware_system_msgs [8.58s]                                                                                                                                                       
Starting >>> autoware_health_checker
Finished <<< ds4_msgs [8.58s]
Finished <<< sick_ldmrs_msgs [8.58s]
Finished <<< autoware_can_msgs [8.59s]
Starting >>> sick_ldmrs_driver
Starting >>> adi_driver
Starting >>> autoware_bag_tools
Finished <<< custom_msgs [8.76s]                                                                                                                                                              
Starting >>> autoware_launcher
Finished <<< autoware_config_msgs [12.5s]                                                                                                                                                              
Starting >>> citysim
--- stderr: ndt_gpu                                                                                                                                                                                                
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
[... 670 lines cut ...]
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[46.283s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[46.283s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[46.283s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[46.283s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[46.283s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[46.283s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[46.283s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[46.283s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[46.283s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[46.283s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[46.283s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_fd2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:220> wait_for=<Future pending cb=[_chain_future.<locals>._call_check_cancel() at /usr/lib/python3.5/asyncio/futures.py:431, Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>

##################

gbiggs, 2019-06-17:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

closed 2019-06-17 by gbiggs

nisanurilhan, 2021-03-24:
How did you solve this error?

cassini-fly, 2021-07-03:
I used the autoware docker image and ade.