autowarefoundation/autoware #2307: op_globle_planner process died during Construct RoadNetwork(V2)

alva786, 2019-06-18:
## Bug report

### Required information:

- Operating system and version:
  - OS and version:Ubuntu 16.04
- Autoware installation type:
  - From source, 
- Autoware version or commit hash
  - tag v.1.11
- ROS distribution and version:
  - Kinetic
- ROS installation type:
  - From source
- Package or library, if applicable:
  -N/A

### Description of the bug
op_globle_planner process died when extracting lanes.

### Steps to reproduce the bug
Map source are from autoware-data : vector_map_sim_sf_portion (or)vector_map_sim_shalun
(1)Load vector map & tf launch
(2)run vel_pose_connect and op_global_planner
(3)See the process has died
[image]
[image]

### Expected behavior
Planning a global path

### Actual behavior
op_globle_planner process died

### Screenshots
[image]

### Additional information
Trace code, it was died in  MappingHelpers.cpp: 
 when doing: ConstructRoadNetworkFromROSMessageV2
by null pointer(AisanLane* pL)  in function IsStartLanePoint
May be it seems that lane.csv format error.???? 
https://github.com/lgsvl/autoware-data/tree/master/data/map/vector_map_sim_shalun/lane.csv
Always crash at when extracting the last laneid.
How can I modify for not process die?
 Is there any specification of generation vector map for normal Construct RoadNetwork_V2?

alva786, 2019-06-19:
Update.
For Shalun Map , if I modify the lane.csv in Shalun maps. At LnID:3965 , modify the FLID from 3966 to 0.
op_globle_planner process "not" died. But the result VectorMap CenterLines is also abnormal like attached picture.
[image]

For loading with SimpleMap,
No need to modify lane.csv, op_globle_planner process "not" died. But the result VectorMap CenterLines is also abnormal like attached picture.
[image]

gbiggs, 2019-06-19:
I think you should ask a question at ROS Answers. You will probably get help faster there. If you then find that it really is a bug, then we can handle it here.

https://github.com/autowarefoundation/Autoware/wiki/Support-guidelines

Fangwq, 2019-09-20:
@alva786 Hi, I also try the method mentioned [here](https://github.com/autocore-ai/MapToolbox/issues/14). The op_global_planner process doesn't die and the map also becomes normal. But the point map and vector map doesn't match any more(you can see the detail there). In addition, when I operate the car in lgsvl simulator, the points raw, and points map also doesn't match. Do you know why that happens and how to fix it? Thank you very much!

closed 2020-05-29 by gbiggs