autowarefoundation/autoware #2308: Way Planner cannot generate global path for lane change case

YaoHsuan, 2019-06-18:
## Bug report

### Required information:

- Operating system and version: OS and version:
  -  Ubuntu 16.04
- Autoware installation type: 
  -  From source
- Autoware version or commit hash  
  -  tag v.1.11
- ROS distribution and version:
  -  Kinetic
- ROS installation type:
  -  from binaries
- Package or library, if applicable:
  - N/A

### Description of the bug

We follow demo steps from youtube to test lane change of path planning. 

	Lane Change of Autoware (https://www.youtube.com/watch?v=C_4rXvKPWMY)
	
According to demo video, way_planner can plan global path for lane change case on moriyama map. We upgrade Autoware to 1.11.0, we try to follow demo video step by step to plan global path with way_planner. But way_planner cannot generate global path with below error message.

PlannerH -> Plan (A) Failed, Trying Plan (B).
Can't Generate Global Path for Start (X:-14770.6000061, Y:-84757.3999996, Z:39, A:2.35619449019
Lon:0, Lat:0, Alt:0, Dir:0) and Goal (X:-14878.6000061, Y:-84713, Z:39, A:-2.73146579822 Lon:0, Lat:0, Alt:0, Dir:0)

Can way_planner support plan path for lane change case on Autoware 1.11.0?

### Steps to reproduce the bug

1. Launch Autoware 1.11.0 , then load vector map from moriyama map folder
[image]

2. Launch vel_pose_connect with "Simulation Mode"
[image]

3. Launch way_planner by command, then assign inital pose with '2D pose estimate' button via RVIZ
[image]

4. Next, assign goal pose of lane change case with '2D Navi Goal' button via RVIZ
[image]

### Expected behavior

Way planner should plan global path for lane change case like demo video
[image]

### Actual behavior

Way planner gets failed to plan global path with error message

[image]

### Screenshots

### Additional information

mitsudome-r, 2019-06-20:
Thanks for using Autoware and for your question. 
However the issue tracker is for posting **confirmed** bugs and feature requests.

> Can way_planner support plan path for lane change case on Autoware 1.11.0?

This question seems to be better asked at the ROS Answers website following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

YaoHsuan, 2019-06-27:
Dear Developers,

Many thanks for your reply. I will report this issue to proper website by your suggestion.
Thanks for your help.

SenRamakri, 2020-05-27:
Hello, I'm seeing the same issue with version 1.13/melodic. Has this been fixed?

closed 2020-05-29 by gbiggs