autowarefoundation/autoware #2312: Gtk warning **: unable to how "app": operation not supported

Jiaminchen0, 2019-06-25:
## Bug report

### Required information:

- Operating system and version:
  - 
Ubuntu 16.04
- Autoware installation type:
  - 
I installed Autoware from source.
- Autoware version or commit hash
  - 

- ROS distribution and version:
  - 
ROS - kinetic
- ROS installation type:
  - 
- Package or library, if applicable:
  - 

### Description of the bug

I am trying to output the PCD file after utilizing NDT mapping, but the system not work. It shows like the screenshot I put here.

### Steps to reproduce the bug

### Expected behavior

### Actual behavior

### Screenshots

[image]
### Additional information

gbiggs, 2019-06-25:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

closed 2019-06-25 by gbiggs

elNiNo009, 2020-03-11:
facing the same probelm

Carolbelieve, 2020-10-20:
do you solved this problem?

Carolbelieve, 2020-10-20:
> facing the same probelm

have you solved this problem?