autowarefoundation/autoware #2315: Auto update points map loader

gnykumar, 2019-07-22:
Hi,

in points_map_loader

514 line

arealist_path += argv[2];

what is that file and what is content it needs?

If possible please someone give me some sample file

gbiggs, 2019-08-04:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://gitlab.com/autowarefoundation/autoware.ai/autoware/wikis/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

closed 2019-08-04 by gbiggs

gnykumar, 2019-08-10:
Hi,

I found how to use this and made a small script to generate
arealist
I used this feature successfully.

Great work from Autoware team