autowarefoundation/autoware #2320: GPU support issue on lidar euclidean cluster

RuiAidriver, 2019-09-17:
Hello,

We are running lidar euclidean cluster as a 3D lidar detection method. When we are using our laptops with a 1050Ti GPU or on an old server with a Tesla K40c. The program is running fine and the volatile of GPU is about 30%.
But when we run the program on the latest 2080Ti, the volatile become 100% and the whole system is extremely slow.
Did anyone face the similar issue? Is this because the latest nvidia GPU doesn't support this algorithm?

Many thanks

seboz123, 2019-10-23:
Facing the same problem. Have you found a solution yet?

RuiAidriver, 2019-10-23:
> Facing the same problem. Have you found a solution yet?

No, I'm using CPU now.

RuiAidriver, 2019-10-23:
@seboz123 We've tried both CUDA 10.0 and CUDA 10.1, none of them is working.

anhnv3991, 2019-10-23:
Thank you for reporting. There are some problems with this method when running on RTX gpu. I am finding the actual problem. In the meantime, please use pascal or older gpu or the cpu clustering.

seboz123, 2019-10-24:
We are are using the nvidia v100 and experiencing the same problem. Did a little bit of debugging. It seems that the 
`gecl_cluster.extractClusters();`  
in lidar_euclidean_cluster_detect.cpp seems to cause the error. In the gpu_euclidean_clustering.cu the line
`buildClusterMatrix << <grid_x, block_x >> > `
still gets executed but the function:
> checkCudaErrors(cudaDeviceSynchronize()); 
never finishes.

anhnv3991, 2019-10-24:
@seboz123 Hmm, I guess the problem comes from the second __syncthreads() in the buildClusterMatrix<<<>>> function. Could you please try to move that __synchtreads() outside of the for loop like this:
    for (int row = 0; row < column; row++)
    {
      float tmp_x = x[row] - local_x[threadIdx.x];
      float tmp_y = y[row] - local_y[threadIdx.x];
      float tmp_z = z[row] - local_z[threadIdx.x];
      int row_cluster = cluster_indices[row];
      int rc_offset = cluster_offset[row_cluster];

      //__syncthreads();   // Comment this out

      if (row_cluster != column_cluster && norm3df(tmp_x, tmp_y, tmp_z) < threshold)
        cluster_matrix[rc_offset * cluster_num + cc_offset] = 1;
    }
    __syncthreads(); // Move it here, right outside the second for loop, but still inside the first for loop

Then compile and run it again and see if it works?

seboz123, 2019-10-25:
This seems to fix the problem! Thank you!

closed 2020-05-29 by gbiggs