autowarefoundation/autoware #2321: ndt_matching not working with Hesai lidar point cloud data

ajay1606, 2019-09-19:
Hello all,
Thanks for Autoware for opensource.

I have been tested Autoware localization (lidar localizer using ndt_matching) with Velodyne (16, 32 and 64) model sensors and Ouster 64 channel sensor. With all the sensor localizer working good. But with the  [Hesai Pandar64 ](https://www.hesaitech.com/en/Pandar64) channel lidar getting an error like this bellow. Above mentioned sensors tested with the same map data and with the frame id modified to "velodyne"

[image]

`For frame [velodyne]: No transform to fixed frame [world].  TF error: [Lookup would require extrapolation into the past.  Requested time 946655660.000381000 but the earliest data is at time 1568785504.549147331, when looking up transform from frame [velodyne] to frame [world]]`

Working Environment
Ubuntu: 16.04
ROS : Kinetic
Autoware version: 1.10.0

Appreciate your help

YamatoAndo, 2019-09-30:
@ajay1606 It seems that TF has not been converted well, because the sensor time stamp is strange. Please modify the sensor driver.

ajay1606, 2019-10-03:
@YamatoAndo , thanks for your reply,  I will test again by modifying the driver and get back to you

duclinhfetel, 2019-10-06:
I'm working on Hesai lidar. I had problems building a map with ndt_mapping. 
@ajay1606  do you have this problem?
[image]

ajay1606, 2019-10-07:
@duclinhfetel In my case, I have changed frame id to **pandar to velodyne**  and tested with the Autoware. It working well with the ndt mapping, detection nodes, point preprocessing nodes. But i have a problem only with the ndt_matching node. Still, I could not fix the issue. 

But you can change frame id, hope ndt_mapping will works fine.

duclinhfetel, 2019-10-07:
> @duclinhfetel In my case, I have changed frame id to **pandar to velodyne** and tested with the Autoware. It working well with the ndt mapping, detection nodes, point preprocessing nodes. But i have [...]

I had a problem when I converted sensor_msgs::pointcloud2 to pcl::PointXYZI. 
Error messages: "Failed to find match for field 'intensity'".
@ajay1606 do you have this problem?

ajay1606, 2019-10-07:
@duclinhfetel Yes i do have the similar message when I converted sensor_msgs::pointcloud2 to pcl::PointXYZI. But that is just a warning message from the PCL library. That should not cause any problem.

duclinhfetel, 2019-10-07:
> @duclinhfetel Yes i do have the similar message when I converted sensor_msgs::pointcloud2 to pcl::PointXYZI. But that is just a warning message from the PCL library. That should not cause any problem.

@ajay1606  I  change line " pcl_conversions::toPCL(ros::Time(timestamp), cld->header.stamp);" in main.cc of hesai lidar to "pcl_conversions::toPCL(ros::Time(ros::Time::now()), cld->header.stamp);" 
fixed your problem. but it don't localization. :)

ajay1606, 2019-10-07:
@duclinhfetel and @YamatoAndo Thanks to both, ndt_matching working now with the following modification
pcl_conversions::toPCL(ros::Time(ros::Time::now()), cld->header.stamp) in the main.cc file of hesai ros package.

duclinhfetel, 2019-10-07:
> @duclinhfetel and @YamatoAndo Thanks to both, ndt_matching working now with the following modification
> pcl_conversions::toPCL(ros::Time(ros::Time::now()), cld->header.stamp) in the main.cc file of hesai ros package.

@ajay1606  do you change original code ndt_matching? ndt_mapping work well, but ndt_matching don't localize :(

ajay1606, 2019-10-07:
@duclinhfetel could you please post the error message?

duclinhfetel, 2019-10-07:
> @duclinhfetel I have not modified the original code, by the way could you please post the error message?

Don't have error message. It don't matching when i use 2D pose estimate in rviz.

ajay1606, 2019-10-07:
@duclinhfetel I guess this could be because of 2 reasons.
1. The initial position is wrong (Please check the initial position again)
2. High point density (Try to use  point down sampling )

Are you able to visualize the point_raw in the RVIZ ?

duclinhfetel, 2019-10-07:
> @duclinhfetel I guess this could be because of 2 reasons.
> 
> 1. The initial position is wrong (Please check the initial position again)
> 2. High point density (Try to use  point down sampling )
> 
> Are you able to visualize the point_raw in the RVIZ ?
> If possible you can check the error message in the generated log file

I found your problem. It's wrong tf_yaw. thanks for support =))

ajay1606, 2019-10-07:
@duclinhfetel cool, thanks to you as well. 
Let's close this issue if you have no more question related to posted issue.

duclinhfetel, 2019-10-07:
@ajay1606  can you give me your email address. May be we will discuss more information about hesai lidar. Thanks. my email: duclinh.fetel@gmail.com

ajay1606, 2019-10-07:
@duclinhfetel Sounds great, please mail me at ajaygkumar16@gmail.com. Thanks

closed 2019-10-07 by ajay1606