autowarefoundation/autoware #2332: Different message types when starting open_planner

jS5t3r, 2020-02-29:
When I start `roslaunch /home/jester/autoware.ai/src/autoware/core_planning/op_local_planner/launch/op_behavior_selector.launch`, I get this error message: 

```
[ERROR] [1582987555.182141549, 7290.582754337]: Client [/rviz_1582987524023091462] wants topic /behavior_state to have datatype/md5sum [visualization_msgs/Marker/4048c9de2a16f4ae8e0538085ebf1b97], but our version has [visualization_msgs/MarkerArray/d155b9ce5188fbaf89745847fd5882d7]. Dropping connection.
```

closed 2020-02-29 by jS5t3r

txxg001001, 2021-04-01:
hi, may I ask how did you fix this issue?  thanks

JWhitleyWork, 2021-04-01:
@txxg001001 If you are having a problem with Autoware.ai that has not been solved, please make a post on [ROS Answers](https://answers.ros.org/) with the `autoware` tag and someone from the community will respond as soon as they are available.

txxg001001, 2021-04-01:
@JWhitleyWork , after update the file default.rviz(change Marker to MarkerArray), this issue can be solved, thanks for your reply.