autowarefoundation/autoware #2333: How to produce pcd file with point type xyz rgb using NDT mapping node?

SLAMgreen, 2020-04-01:
Based on rosbag named sample_moriyama_150324.bag, I used to create a pcd map by using NDT mapping node.
But the point type of  pcd map by myself  is xyz intensity.The pcd file autoware provided in the quick start is xyz rgb.It is different ! 
So could you please tell me how to produce the pcd files which point types are xyz rgb?

SLAMgreen, 2020-04-02:
If the NDT mapping node create a cloud map with point type xyzi,could you please tell me what is the method to transform the xyzi to xyzrgb?

Pallav1299, 2020-04-25:
First of all you would need pointcloud with XYZRGB data. You need to get the RGB data from camera feed and fuse it with LIDAR's XYZ data. Check the Autoware's Pixel_cloud_fusion for the same. This will generate an output pointcloud with XYZRGB data. 
Possible changes in the NDT mapping node should be, changing the point types to pcl::PointXYZRGB wherever required. Also remember to change the topic names if required.

Pallav1299, 2020-04-28:
> First of all you would need pointcloud with XYZRGB data. You need to get the RGB data from camera feed and fuse it with LIDAR's XYZ data. Check the Autoware's Pixel_cloud_fusion for the same. This [...]

I've tried this. Just needed to change the point-type. Also "anh" method was not supported to XYZRGB so had to comment all instances of "anh". I don't have much knowledge about "anh". If someone can explain the reason, it'll be helpful.

closed 2020-05-29 by gbiggs