autowarefoundation/autoware #2338: Pose error when using NDT matching with UTM coordinates

umiklos, 2020-06-29:
## Bug report

### Required information:

- Operating system and version:
  -  Ubuntu 18.04
- Autoware installation type:
  -  From source
- Autoware version or commit hash
  - 1.13
- ROS distribution and version:
  -  Melodic
- ROS installation type:
  -  From source
- Package or library, if applicable: 
  -  Lidar localizer

### Description of the bug

We use NDT matching node for localization and we use our GNSS system with UTM coordinates. When we use the ndt_matching node with GNSS we get quadratic errors for topic /ndt_pose. We found out that this cased because you use float instead of double for translation In ndt_matching.cpp at line 1017.  

### Expected behavior

We would like to get a smooth curve like our gnss
[image]

### Actual behavior

[image]

### Screenshots

### Additional information

JWhitleyWork, 2020-07-07:
@umiklos Would it be possible for you to submit a Pull Request which resolves this issue?

xczhanjun, 2021-12-12:
I would suggest creating a local map frame somewhere near the location where you actually drive and use that frame in ndt_matching to do localization.==great