autowarefoundation/autoware #2339: which autoware version have calibrationtool?

Jihun-Kim-kmu, 2020-07-07:
I installed autoware.ai 1.12.0 (master version) but couldn't find calibrationtool. 
There is only calibration publisher.
which autoware version have calibrationtool butoon>

taichiH, 2020-07-07:
@Jihun-Kim-kmu 
please refer to https://github.com/Autoware-AI/utilities/tree/master/autoware_camera_lidar_calibrator#how-to-calibrate

Jihun-Kim-kmu, 2020-07-07:
I installed autoware1.12.0 version. And entered the code using the link.( https://github.com/Autoware-AI/utilities/tree/master/autoware_camera_lidar_calibrator#how-to-calibrate)
/home/kaai/Pictures/dd.png
but error occur
/home/kaai/Pictures/runerror.png
Do I need additional package installation?

taichiH, 2020-07-07:
`SQUARE_SIZE` is OptionParser parameter, you need to set checker board square size.

If you have never tried camera calibration, this link may help your understand.
http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration

Jihun-Kim-kmu, 2020-07-07:
I understand about SQUAR_SIZE.
But doesn't this error mean that haven't found the package?

taichiH, 2020-07-07:
can you find package by this command?
```
rospack find autoware_camera_lidar_calibrator
```

please check if workspace properly sourced.

Jihun-Kim-kmu, 2020-07-07:
[image]
I installed it by referring to this site.(https://github.com/Autoware-AI/autoware.ai/wiki/Source-Build)

taichiH, 2020-07-08:
please source your workspace. 
http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment

Jihun-Kim-kmu, 2020-07-08:
[image]

Jihun-Kim-kmu, 2020-07-09:
I succeeded in calibration the camera.
[image]
Next I have to lidar and camera calibration
Should I copy this site?  (https://github.com/ankitdhall/lidar_camera_calibration#fusion-using-lidar_camera_calibration)

taichiH, 2020-07-09:
great!

> Should I copy this site? (https://github.com/ankitdhall/lidar_camera_calibration#fusion-using-lidar_camera_calibration)

Although I've never tried this way, I think you can calibrate camera lidar calibration using this repository.

The intrinsic parameter file that you create is a widely used format in ROS, but the format of autoware.ai is a bit different (yaml key name is different ([different part](https://github.com/Autoware-AI/utilities/blob/62bcb83daa9f3b092087063cba46544ebeda04e0/autoware_camera_lidar_calibrator/src/autoware_camera_lidar_calibration/camera_lidar_calibration_node.cpp#L128-L133))), so if you try autoware.ai node, you need to calibrate camera intrinsic parameter using autoware.ai node.

Jihun-Kim-kmu, 2020-07-09:
I want to succeed  calibration in autoware.ai.
but I still can't use autoware_camera_ lidar_calibrator package.
How can i use  autoware_camera_ lidar_calibrator package.
I really want to know
[image]

taichiH, 2020-07-09:
please show me the path where autoware_camera_lidar_calibrator is.

Jihun-Kim-kmu, 2020-07-09:
[image]
[image]

taichiH, 2020-07-09:
Did you try this command on autoware.ai root directory?
```
source install/setup.bash
```

Jihun-Kim-kmu, 2020-07-09:
Whenever I run this program, I do this.
cd autoware.ai
source install/setup.bash
roslaunch runtime_manager runtime_manager.launch

taichiH, 2020-07-09:
```
cd autoware.ai
source install/setup.bash
rosrun autoware_camera_lidar_calibrator cameracalibrator.py --square SQUARE_SIZE --size MxN image:=/image_topic
```

Jihun-Kim-kmu, 2020-07-09:
/home/kaai/Pictures/ 타치야 제발.png
but can't find cheesbooard.  Do i have to do install opencv?

Jihun-Kim-kmu, 2020-07-09:
[image]

but can't find cheesbooard. Do i have to do install opencv?

Jihun-Kim-kmu, 2020-07-09:
I succeeded calibration in autoware. ほんとうに、ありがとうございます。
[image]

taichiH, 2020-07-14:
@Jihun-Kim-kmu Can you close the issue?

Jihun-Kim-kmu, 2020-07-14:
sorry i have more question.
Can I close it after camera lidar calibration finish ?

Jihun-Kim-kmu, 2020-07-14:
I succeeded camera calibration in autoware and get autoware camerea calibration.yaml
Next step  I ran camera  and lidar 
[image]

Launch calibration camera and lidar 
roslaunch autoware_camera_lidar_calibrator camera_lidar_calibration.launch intrinsics_file:=/home/kaai/20200713_1843_autoware_camera_calibration.yaml image_src:=/image_raw
[image]
Lines from the lidar sensor do not appear.
Is there anything wrong with the execution process?

Jihun-Kim-kmu, 2020-07-14:
I succeeded camera calibration  and get autoware camera calibration yaml
I played camera and lidar sensor and calibration
roslaunch autoware_camera_lidar_calibrator camera_lidar_calibration.launch intrinsics_file:=/home/kaai/20200713_1843_autoware_camera_calibration.yaml image_src:=/image_raw
[image]
[image]
But In rviz image topic is appear but poin topic don't apper
Is there anything wrong with my process?

JWhitleyWork, 2020-07-16:
@Jihun-Kim-kmu While we want to help as much as possible, support requests should be made on https://answers.ros.org/ with the tag `autoware` instead of here. Issues on these repositories are for confirmed bugs or feature requests only. Please see our [Support Guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines) for more information.

closed 2020-07-16 by JWhitleyWork