autowarefoundation/autoware #2340: Cannot start runtime manager

jS5t3r, 2020-07-07:
My system is Ubuntu 18. ROS Melodic, Cuda 10, tensorrt 5.1

```
roslaunch runtime_manager runtime_manager.launch

Process Manager
# Option “--command” is deprecated and might be removed in a later version of gnome-terminal.
# Use “-- ” to terminate the options and put the command line to execute after it.
```

tekrajchhetri, 2020-07-10:
@jitrc  what version of autoware are you using? Its working fine with ubuntu 18 and autoware 1.12.0 for me.

jitrc, 2020-07-12:
I think @tekrajchhetri you wanted to tag @jS5t3r

tekrajchhetri, 2020-07-13:
Yes, indeed and thank you for correction @jitrc.

JWhitleyWork, 2020-07-16:
@jS5t3r Where did `zshrc` come in? Also, you have to source at least one ROS workspace before using any part of Autoware.AI. Additionally, this is not a confirmed bug and, as such, should be posted at https://answers.ros.org/ instead.

closed 2020-07-16 by JWhitleyWork

jS5t3r, 2020-07-18:
@tekrajchhetri  Ubuntu 18 and AW 1.14

@JWhitleyWork when I use .bashrc it works fine...

jS5t3r, 2020-07-22:
Solved, because this one is solved: https://github.com/carla-simulator/carla-autoware/issues/84