autowarefoundation/autoware #2341: Unable to stop at stop line (Vector maps) using open planner

tekrajchhetri, 2020-07-09:
Hi,

I am able to simulate open planner using both autoware and lgsvl simulator but it is not stopping at the stop lines. I have created an support in ros wiki [https://answers.ros.org/question/356722/unable-to-stop-at-stop-line-vector-maps-using-open-planner/](https://answers.ros.org/question/356722/unable-to-stop-at-stop-line-vector-maps-using-open-planner/) . @hatem-darweesh could you have a look into the issue?

Thank you.

JWhitleyWork, 2020-07-16:
@tekrajchhetri I'm sorry but I do not have enough information to help you out here. Please provide me with the following:

- Operating System (and version):
- Version or commit hash of Autoware being used:
- ROS Installation Method (source or binaries):
- Environment (Docker or native):
- List of all nodes being used:
- Output of `rostopic list` and `rosnode list`

JWhitleyWork, 2020-07-19:
It looks like you're getting plenty of help on ROS Answers, which is where this type of question belongs. Closing.

closed 2020-07-19 by JWhitleyWork