autowarefoundation/autoware #2345: MappingHelpers.cpp Line 3396 code error

dufengit, 2020-07-30:
## Bug report

### Required information:
code error

### Description of the bug

              if(pWP2->pRight == 0)
              {
                pWP2->pRight = pWP->pLeft;
                pWP2->RightPointId = pWP->id;
                pWP2->RightLnId = pL->id;
                pL2->pRightLane = pL;
              }
MappingHelpers.cpp Line 3396: 
pWP2->pRight = pWP->pLeft;
should be:
pWP2->pRight = pWP;

dufengit, 2020-07-31:
> Send time: Friday, Jul 31, 2020 1:12 AM
>
>
> @dufengit What you've described here is not a bug but a change to the code with no explanation. In addition, unconfirmed bugs should first be posted to ROS Answers with the autoware tag.

In fact, it is a BUG, I debug the code, and find this BUG.
If using the original code, for a 2 lane map, code in line 3396 will make the point on the left lane's right point point to itself.
(1) Look at code section above, you may find they are similar. 
(2) And if you assign pWP-&gt;pLeft to pWP2-&gt;pRight , then why pWP-&gt;id
is assigned to pWP2-&gt;RightPointId, not pWP2-&gt;RightPointId = pWP-&gt;pLeft-&gt;id, this doesn't make sense.

dufengit, 2020-07-31:
In fact, it is a BUG, I debug the code, and find this BUG.
If using the original code, for a 2 lane map, code in line 3396 will make the point on the left lane's right point point to itself.
(1) Look at code section above, you may find the code is similar.
(2) And if you assign pWP->pLeft to pWP2->pRight , then why pWP->id
is assigned to pWP2->RightPointId, not pWP2->RightPointId = pWP->pLeft->id, this doesn't make sense.

              if(pWP->pRight == 0)
              {
                pWP->pRight = pWP2;
                pWP->RightPointId = pWP2->id;
                pWP->RightLnId = pL2->id;
                pL->pRightLane = pL2;

              }

              if(pWP2->pLeft == 0)
              {
                pWP2->pLeft = pWP;
                pWP2->LeftPointId = pWP->id;
                pWP2->LeftLnId = pL->id;
                pL2->pLeftLane = pL;
              }
            }
            else
            {
              if(pWP->pLeft == 0)
              {
                pWP->pLeft = pWP2;
                pWP->LeftPointId = pWP2->id;
                pWP->LeftLnId = pL2->id;
                pL->pLeftLane = pL2;
              }

              if(pWP2->pRight == 0)
              {
                pWP2->pRight = pWP->pLeft;
                pWP2->RightPointId = pWP->id;
                pWP2->RightLnId = pL->id;
                pL2->pRightLane = pL;
              }
            }

JWhitleyWork, 2020-07-31:
@dufengit My point was that you didn't report a bug, you just stated "this is a bug" without any context or examples. You used the "Bug Report" issue type but you completely deleted the contents of the template which asks:

```
- Operating system and version:
  - <!-- OS and version (e.g. Ubuntu 18.04, MacOS 10.15, Windows 10 build 1817) -->
- Autoware installation type:
  - <!-- How did you install Autoware? From source, from binaries, Docker, etc. Link to a guide if you followed one. -->
- Autoware version or commit hash
  - <!-- If from binaries or docker, give the version. If from source, give the output of git rev-parse HEAD or the repos file you use -->
- ROS distribution and version:
  - <!-- State the name of the ROS distribution you are using, and if applicable a patch version -->
- ROS installation type:
  - <!-- How did you install ROS? From source, from binaries, Docker, etc. Link to a guide if you followed one. -->
- Package or library, if applicable:
  - <!-- e.g. velodyne_driver, or N/A -->
```

as well as the **Steps to Reproduce the Bug** section. You must fill out **all** of these fields if you expect to receive help. Please open a new Issue and fill out the template.

dufengit, 2020-08-01:
Thanks for your time
 &nbsp;
 - Autoware version or commit hash
 I clone the code in july (2020.7), today i clone again and find there is no src code in the github repository.
 For the other information, i think it is not relevant, just imagine white-box testing.
 &nbsp;
 I will not open new issue, just save time for both of us. 
 Thanks
  

 ------------------ Original ------------------
  From: &nbsp;"Joshua Whitley";<notifications@github.com&gt;;
 Send time:&nbsp;Saturday, Aug 1, 2020 5:25 AM
 To:&nbsp;"Autoware-AI/autoware.ai"<autoware.ai@noreply.github.com&gt;; 
 Cc:&nbsp;"dufen"<295735116@qq.com&gt;; "Mention"<mention@noreply.github.com&gt;; 
 Subject: &nbsp;Re: [Autoware-AI/autoware.ai] MappingHelpers.cpp Line 3396 code error (#2345)

 

 

 
@dufengit My point was that you didn't report a bug, you just stated "this is a bug" without any context or examples. You used the "Bug Report" issue type but you completely deleted the contents of the template which asks:
- Operating system and version:   -

hatem-darweesh, 2020-08-02:
@dufengit Yes, it is a bug when you try to use the lane change objection. 
Unless you have a very complex road network map, this will not be an issue. 
nothing will crash. 
if you don't have time to go through all the process of reporting, explaining, and MR just modify it in your local repository. 
Thank you for mentioning the problem here.