autowarefoundation/autoware #2351: ndt_mapping not working properly!

AswinBBT, 2020-10-20:
While selecting ndt_mapping from the runtime manager in the computing tap ,showing error in the terminal as 
"**RLException: [ndt_mapping.launch] is neither a launch file in package [lidar_localizer] nor is [lidar_localizer] a launch file name
The traceback for the exception was written to the log file**" 
Steps Followed are :
1.Go to Autoware/ros directory
2.Run Autoware using “./run” command
3.Go to Simulation tab and Load a ROSBAG
4.Click Play and immediately PAUSE
5.Click Computing tab and select ndt_mapping
6.Click RViz button at the bottom
7.In Rviz click File menu and then click Open Config to select visualization template for ndt_mapping.rviz located in Autoware/src/.config/rviz7 
8.ndt_mapping will read from /points_raw
IF the pointcloud is being published in a different topic, use the relay tool in a new terminal window
rosrun topic_tools relay /front/velodyne_points /points_raw
This will forward the topic /front/velodyne_points to /points_raw
9. Go back to Simulation tab and click Pause to start mapping
10. Mapping process can be seen from Rviz
[image]

kosmastsk, 2020-10-20:
Was the workspace built  successfully?

AswinBBT, 2020-10-21:
> Was the workspace built successfully?
Autoware workspace compilation was not completed, some more packages are there to complete.
Is that the issue for this error ?

AswinBBT, 2020-10-21:
@kosmastsk Autoware workspace compilation was not completed, some more packages are there to complete.
Is that the issue for this error ?

kosmastsk, 2020-10-21:
You should make sure that at least the lidar_localizer package has been properly built

AswinBBT, 2020-10-30:
I have built almost 70 packages, lidar_localizer is not included in it
[image]
The above error (calibration_publisher) occurs after building 70 packages . Any idea for removing this error?

kosmastsk, 2020-10-30:
@AswinBBT this looks like an OpenCV error, wrong linkage between your workspace and the OpenCV installation. Cannot tell more from this screenshot, however in order to run the ndt_mapping you have to make sure that the package has been built. Make sure that you have the correct versions of ROS, autoware, CUDA, OpenCV and all the other pre-requisites and try again.

AswinBBT, 2020-10-30:
Ok fine 😊👍, CUDA is optional right ? Think CUDA is not mandatory for ndt_mapping ? Any idea

kosmastsk, 2020-10-30:
No, it's not mandatory. Only if you plan to use the ndt_gpu functionalities, otherwise it runs fine without it

AswinBBT, 2020-10-30:
Ok thank you 😊

AswinBBT, 2020-10-30:
One more doubt regarding the lidar , we are using data from RpLidar A3 for this and it is a 2D Lidar , but in autoware is it possible to use this ?
[image]

From the list of Lidars in autoware there is no RpLidar A3 and Is it needed to use data of any of the lidar from this list itself ?

kosmastsk, 2020-10-30:
If you make sure that your sensor is publishing at the same topic as the sensors available in the list, you should not have any issue.

JWhitleyWork, 2020-10-30:
Autoware.AI was not designed or tested with 2D lidar. Additionally, you will usually get a `sensor_msgs/LaserScan` from a 2D lidar driver and a `sensor_msgs/PointCloud2` from a 3D lidar, which is what the Autoware.AI code was designed to handle.

AswinBBT, 2020-10-31:
@JWhitleyWork Ok.. But here we convert our LaserScan data to PointCloud2 inorder to test with the autoware ? Is that possible

JWhitleyWork, 2020-11-17:
I am not sure but someone else in the community is likely to be able to answer. Per our [support guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines), please submit support questions to [ROS Answers](https://answers.ros.org/).

closed 2020-11-17 by JWhitleyWork