autowarefoundation/autoware #2353: parameters in dp_planner

MINUS-1ONE, 2020-10-23:
I can not find any explanation about parameter in dp_planner, such as `horizontalSafetyDistance`, `verticalSafetyDistance`, `minDistanceToAvoid`, `maxDistanceToAvoid` and so on, and I got bad performance in obstacles avoidance by using default value of parameter. So I sincerely hope that there is a explanation about the parameters in dp_planner. Thanks in advance.

SenRamakri, 2020-10-23:
@MINUS-1ONE - Have you tried looking into this document - https://www.fujipress.jp/jrm/rb/robot002900040668/?
This paper explains how the algorithm works and may answer your questions. Also look at Tables 3, 5 in that document which provides some info on the parameters.

MINUS-1ONE, 2020-10-23:
> @MINUS-1ONE - Have you tried looking into this document - https://www.fujipress.jp/jrm/rb/robot002900040668/?
> This paper explains how the algorithm works and may answer your questions. Also look at Tables 3, 5 in that document which provides some info on the parameters.

thank you for your reply! I am checking that paper and found a parameter `enableFollow` that enable the car to follow something, which is useful for me.

closed 2020-11-17 by JWhitleyWork