autowarefoundation/autoware #2356: runtime manager cannot be launch (with process died)

llu0120, 2020-11-12:
I'm currently using master branch. 
With Ubuntu 18.04 and ROS melodic installed, after I start/enter the Autoware docker with `./run.sh` and I tried to launch the `runtime manager`. However, when I do `roslaunch runtime_manager runtime_manager.launch`, error occurs with process died like below. 

`... logging to /home/autoware/.ros/log/37d2e086-24f5-11eb-808d-c400ad2bc16a/roslaunch-axcar01-77.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://axcar01:37285/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.6

NODES
  /
    run (runtime_manager/run)

auto-starting new master
process[master]: started with pid [87]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 37d2e086-24f5-11eb-808d-c400ad2bc16a
process[rosout-1]: started with pid [98]
started core service [/rosout]
process[run-2]: started with pid [101]
Unable to init server: Could not connect: Connection refused
[run-2] process has died [pid 101, exit code 1, cmd /home/autoware/Autoware/install/runtime_manager/share/runtime_manager/scripts/run __name:=run __log:=/home/autoware/.ros/log/37d2e086-24f5-11eb-808d-c400ad2bc16a/run-2.log].
log file: /home/autoware/.ros/log/37d2e086-24f5-11eb-808d-c400ad2bc16a/run-2*.log
`

the log file `run-2*.log` shows error like below: 
`Process Manager
# Option “--command” is deprecated and might be removed in a later version of gnome-terminal.
# Use “-- ” to terminate the options and put the command line to execute after it.
# Failed to parse arguments: Cannot open display: 
`

llu0120, 2020-11-12:
I believe it's a problem of visualizing the runtime manager on my `local laptop` for a `remote host`. Please help thanks!

JWhitleyWork, 2020-11-17:
@llu0120 Per our [support guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines), please submit support questions to [ROS Answers](https://answers.ros.org/).

closed 2020-11-17 by JWhitleyWork