autowarefoundation/autoware #2357: ndt_mapping doesn't compute transformation matrix

mix345, 2020-11-12:
Hi,

I run ndt_mapping on the demo bag file (sample_moriyama_150324.bag). It shows /point_raw on rviz, but /points_map can't be visualized and /current_pose is always zero matrix. Does anyone have an idea why?

**Environment**: Ubuntu18 and Autoware master version.

**Steps**: 

1. Download a demo bag files and setup Autoware master version (https://github.com/Autoware-AI/autowa...)
2. Launch runtime_manager
3. Launch tf of velodyne and map
4. Launch ndt_mapping
5. Launch Rviz
6. Cancel the pause of .bag file

**Console output:** 

'-----------------------------------------------------------------
(Processed/Input): (306 / 308)
'-----------------------------------------------------------------
Sequence number: 86918
Number of scan points: 39581 points.
Number of filtered scan points: 3091 points.
transformed_scan_ptr: 39581 points.
map: 39928 points.
NDT has converged: 1
Fitness score: 0
Number of iteration: 0
(x,y,z,roll,pitch,yaw):
(0, 0, 0, 0, -0, 0)
Transformation Matrix:
0 0 0 0
0 0 0 0
0 0 0 0
0 0 0 0
shift: 0
'-----------------------------------------------------------------

JWhitleyWork, 2020-11-17:
@mix345 Per our [support guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines), please submit support questions to [ROS Answers](https://answers.ros.org/).

closed 2020-11-17 by JWhitleyWork

Kin-Zhang, 2021-03-06:
I also met this problem on v1.14 And I found someone post on ROS Answers but nobody answers this, the host change the version to v1.13 and it works. Here is the link: https://answers.ros.org/question/365415/autoware-ndt_mapping-doesnt-compute-transformation-matrix/