autowarefoundation/autoware #2358: obstacle avoid with Autoware 1.12.0 open_planner

c1hong, 2020-11-25:
Hi, I have some problem. 
I want to avoid obstacle with autoware, but my car is breaking through the obstacle.
I have vector map(dtlane), points_raw, current_pose.
So, I didn't launch 'ndt_matching'.

I launched these below nodes.

- vector_map loader
- TF (world to map, map to mobility, baselink to localizer)
- ray_ground_filter
- lidar_euclidean_cluster_detect
- imm_ukf_pda_track
- naive_motion_predict
- op_global_planner
- op_common_params
- op_trajectory_generator
- op_trajectory_evaluator
- op_behavior_selector
- pure_pursuit
- twist_filter

This picture is the current my situation.

[image]

In the photo, the color of the roll_out does not appear 'RED' even though there are obstacles in the driving path.

How can I solve this problem? 
I already ask at ROS Answers, but there's no answer to this question.. 
I look forward to hearing from you in the near future.
Thank you. :)

JWhitleyWork, 2020-12-02:
@c1hong Can you please link to the ROS Answers question? I will see if I can get some help there. This is not a good place for visibility for others in the community..

closed 2020-12-02 by JWhitleyWork

c1hong, 2020-12-03:
https://answers.ros.org/question/365289/how-to-implement-avoid-obstacle-with-open_planner-in-autoware/
This is the link.