autowarefoundation/autoware #2363: Not able to run Demo using the docker file.

harendracmaps, 2020-12-23:
## Bug report

### Required information:

- Operating system and version:
  Ubuntu 18.04
- Autoware installation type:
Docker 
- Autoware version or commit hash
autoware/autoware:latest-melodic        90e973a21691        5 months ago.  (1.14.0 version got from dockerhub)
- ROS distribution and version:
1.14.6
- ROS installation type:
Docker as above
- Package or library, if applicable:
Package

### Description of the bug
I am not able to run demo and get pointcloud.

### Steps to reproduce the bug

1. git clone https://github.com/Autoware-AI/autoware.ai.git
2. cd docker/generic
3. ./run.sh -c off -s
4. roslaunch runtime_manager runtime_manager.launch
5. Followed steps given https://github.com/Autoware-AI/autoware.ai/wiki/ROSBAG-Demo
6. Attached the screen shots about the results. not able to run demo as shown in the link.

### Expected behavior
1. Followed steps given https://github.com/Autoware-AI/autoware.ai/wiki/ROSBAG-Demo
2. Attached the screen shots about the results. not able to run demo as shown in the link.

### Actual behavior
Some time point is moving but road and lan marking are not visible pointcloud is also not getting rendered.

### Screenshots

[image]
[image]
[image]

### Additional information

JWhitleyWork, 2021-01-04:
@harendracmaps Based on our [support guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines), unconfirmed bugs and support requests should be made on [ROS Answers](https://answers.ros.org/) with the `autoware` tag. Please re-post there.

closed 2021-01-04 by JWhitleyWork

powerpzt, 2021-04-16:
Hi. I have same problem to you.
Actually, the latest-melodic docker version has 14.0 version, which is different to master version.
If you can use master version, download the bleedingedge (master) version docker.
Because the docker container is base version, you need to compile.

$ docker pull autoware/autoware:bleedingedge-melodic-base-cuda
$ docker run ~~blahblah~~ autoware/autoware:bleedingedge-melodic-base-cuda
(I use the run command from docker.git/generic/run.sh)

In docker, I followed the compile from https://github.com/Autoware-AI/autoware.ai/wiki/Source-Build

$ git clone https://github.com/Autoware-AI/autoware.ai.git
$ mkdir -p autoware.ai/src
$ cd autoware.ai
$ wget -O autoware.ai.repos "https://raw.githubusercontent.com/Autoware-AI/autoware.ai/master/autoware.ai.repos"
$ vcs import src < autoware.ai.repos
$ rosdep update
$ rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
$ AUTOWARE_COMPILE_WITH_CUDA=1 colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release