autowarefoundation/autoware #2366: ndt_gpu too many resources requested for launch

Bruce-Cui, 2021-02-01:
I use ndt_gpu lib and load my pointcloud map with `void setInputTarget(pcl::PointCloud<pcl::PointXYZ>::Ptr input)`,
then I found if I change the resolution param or change the BLOCK_SIZE_X, sometimes I would get a error "Error: too many resources requested for launch ", but sometimes it looks ok.
the error would happens in VoxelGrid.cu setInput() when runs to computeCentroidAndCovariance() or buildOctree() function.
I run it in Ubuntu 18.04 and GTX 1660 Ti.

[image]

```//Input are supposed to be in device memory
void GVoxelGrid::setInput(float *x, float *y, float *z, int points_num)
{
	if (points_num <= 0)
		return;
	x_ = x;
	y_ = y;
	z_ = z;
	points_num_ = points_num;

	findBoundaries();

	voxel_num_ = vgrid_x_ * vgrid_y_ * vgrid_z_;

	initialize();

	scatterPointsToVoxelGrid();

	computeCentroidAndCovariance();

	buildOctree();
} ```

Bruce-Cui, 2021-02-02:
I can run ndt_gpu on GTX1050 and GTX1060，but fails on GTX1660Ti.
this is GTX1060.
[image]

JWhitleyWork, 2021-02-23:
Please see our [Support Guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines) for how to get support. The first step is creating a question on [ROS Answers](https://answers.ros.org/) with the `autoware` tag.

closed 2021-02-23 by JWhitleyWork