autowarefoundation/autoware #2369: autoware colcon build and exec runtime_manager problem

stevezhou7020, 2021-04-15:
**issue1:**
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> autoware_build_flags
[0.937s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/ros/autoware.ai/devel' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> autoware_msgs
Starting >>> vector_map_msgs
Starting >>> autoware_lanelet2_msgs
Starting >>> autoware_config_msgs
Starting >>> ros_observer
Starting >>> autoware_system_msgs
Starting >>> autoware_can_msgs
Finished <<< autoware_build_flags [1.14s]                                  
Starting >>> gnss
Finished <<< autoware_lanelet2_msgs [1.32s]
Starting >>> tablet_socket_msgs
Finished <<< autoware_can_msgs [1.70s]                                  
Starting >>> sick_ldmrs_description
Finished <<< autoware_system_msgs [1.82s]                               
Starting >>> sick_ldmrs_msgs
Finished <<< ros_observer [2.33s]                                          
Starting >>> autoware_health_checker
Finished <<< autoware_config_msgs [2.44s]                                   
Starting >>> ds4_msgs
Finished <<< gnss [1.62s]                                                    
Starting >>> custom_msgs
Finished <<< sick_ldmrs_description [1.11s]                                  
Starting >>> ds4_driver
Finished <<< tablet_socket_msgs [1.66s]                                      
Starting >>> gazebo_camera_description
Finished <<< autoware_msgs [3.04s]                                           
Finished <<< vector_map_msgs [3.06s]
Starting >>> amathutils_lib
Starting >>> vector_map
Finished <<< sick_ldmrs_msgs [1.31s]                              
Starting >>> astar_search
Finished <<< ds4_msgs [1.21s]                                       
Starting >>> sick_ldmrs_driver
Finished <<< ds4_driver [0.84s]
Starting >>> gazebo_imu_description
Finished <<< gazebo_camera_description [0.75s]
Starting >>> ndt_cpu
Finished <<< custom_msgs [1.44s]        
Starting >>> ndt_tku
Finished <<< gazebo_imu_description [0.80s]                                   
Starting >>> pcl_omp_registration
Finished <<< ndt_cpu [0.94s]                                                  
Starting >>> fastvirtualscan
Finished <<< vector_map [1.84s]                                       
Starting >>> vector_map_server
Finished <<< ndt_tku [0.77s]                                          
Starting >>> kitti_box_publisher
Finished <<< pcl_omp_registration [0.84s]                                    
Starting >>> kitti_player
Finished <<< amathutils_lib [2.61s]                                 
Starting >>> lanelet2_extension
Finished <<< fastvirtualscan [1.27s]                                      
Starting >>> libwaypoint_follower
Finished <<< sick_ldmrs_driver [2.34s]
Starting >>> ndt_gpu
Finished <<< autoware_health_checker [3.81s]                              
Starting >>> rosinterface
Finished <<< astar_search [3.11s]                                         
Starting >>> state_machine_lib
Finished <<< kitti_box_publisher [1.44s]                                  
Starting >>> vehicle_sim_model
--- stderr: lanelet2_extension                                            
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "lanelet2_core"
  with any of the following names:

    lanelet2_coreConfig.cmake
    lanelet2_core-config.cmake

  Add the installation prefix of "lanelet2_core" to CMAKE_PREFIX_PATH or set
  "lanelet2_core_DIR" to a directory containing one of the above files.  If
  "lanelet2_core" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:26 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "lanelet2_core"
  with any of the following names:

    lanelet2_coreConfig.cmake
    lanelet2_core-config.cmake

  Add the installation prefix of "lanelet2_core" to CMAKE_PREFIX_PATH or set
  "lanelet2_core_DIR" to a directory containing one of the above files.  If
  "lanelet2_core" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:26 (find_package)

---
Failed   <<< lanelet2_extension [0.77s, exited with code 1]
Aborted  <<< kitti_player [1.37s]                  
Aborted  <<< vector_map_server [1.86s]                                         
Aborted  <<< rosinterface [1.59s]                                              
Aborted  <<< vehicle_sim_model [1.37s]
Aborted  <<< state_machine_lib [1.49s]
Aborted  <<< ndt_gpu [1.80s]
Aborted  <<< libwaypoint_follower [9.19s]
                                    
Summary: 27 packages finished [16.0s]
  1 package failed: lanelet2_extension
  7 packages aborted: kitti_player libwaypoint_follower ndt_gpu rosinterface state_machine_lib vector_map_server vehicle_sim_model
  2 packages had stderr output: lanelet2_extension ndt_gpu
  119 packages not processed
issue 2:
ros@ros:~/autoware.ai$ source install/setup.bash 
ros@ros:~/autoware.ai$ roslaunch runtime_manager runtime_manager.launch
[runtime_manager.launch] is neither a launch file in package [runtime_manager] nor is [runtime_manager] a launch file name
The traceback for the exception was written to the log file.
thank you for your respose.

abhishekt711, 2021-05-01:
I am also facing the issue in colcon build. Pls help.
[image]

JWhitleyWork, 2021-05-01:
Per our [Support Guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines), please post support requests and unconfirmed bugs to [ROS Answers](https://answers.ros.org) with the tag `autoware`.

closed 2021-05-01 by JWhitleyWork