autowarefoundation/autoware #2377: Runtime managere の画面がぐちゃぐちゃになってしまいます。

konnoeric, 2021-05-19:
このようにhokuyo3dのドライバを起動してスクロールしたりRuntime　Manager内の他のタブを開いたりすると字が重なりグチャグチャになってしまいます。何が問題でしょうか。
[image]

以下に実行環境をスクリーンショットで示します。他にも必要な情報があれば申し付けてください。

↓Autoware.aiを実行したとき
[image]
↓使用しているNvidiaGPUのグラフィック環境
[image]
↓パソコンのスペック
[image]

## Bug report

### Required information:

- Operating system and version:
  - 
- Autoware installation type:
  - 
- Autoware version or commit hash
  - 
- ROS distribution and version:
  - 
- ROS installation type:
  - 
- Package or library, if applicable:
  - 

### Description of the bug

### Steps to reproduce the bug

### Expected behavior

### Actual behavior

### Screenshots

### Additional information

s95huang, 2021-05-21:
You have CUDA 11.3 based on your nvdia-smi
Please check your cuda runtime version using nvcc -V
It seems like you are running Autoware on cuda 11.3, which is not supported
You would need to install CUDA 10.0

Your nvcc -V should look like or similar to:
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2018 NVIDIA Corporation
Built on Sat_Aug_25_21:08:01_CDT_2018
Cuda compilation tools, release 10.0, V10.0.130

JWhitleyWork, 2021-05-22:
@s95huang `nvidia-smi` does not show the installed version of CUDA, unfortunately, it shows the version of CUDA that the driver is compatible with.

Regardless of this, though, unconfirmed bugs and support requests should first be handled through [ROS Answers](https://answers.ros.org) per our [Support Guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines).

closed 2021-05-22 by JWhitleyWork