autowarefoundation/autoware #2379: Perception: COSTMAP Fails to generate map | Planning: module fails to accomplish evasive Maneuver

marcusvinicius178, 2021-05-31:
###Operating system and version:
- Hi I am using Ubuntu 18.04, Ros Melodic, Autoware.AI Version 1.14.0 on Docker environment. I followed this tutorial https://www.svlsimulator.com/docs/system-under-test/autoware-instructions/

### Description of the bug

**The Occupancy_grid Mapping does not update the map properly** as you can check here:
https://www.youtube.com/watch?v=9gbq60B6Yb0&list=PL_d_YiRmUA3nnTRmF8QJKyPoEQhrdvpP2&index=23
And as consequence the motion planner rarely draws the trajectory correctly to avoid the obstacle. I followed several different tutorials that are supposed to build a COSTMAP:

**1 - Launch Costmap_Generator to enable the Occupancy_grid_map topic:**
Tutorial followed: https://gitlab.com/autowarefoundation/autoware.ai/core_planning/tree/9a3856cc3097cfb2c43db6d9c3083589d658e35b/costmap_generator

**Attempts and Results:**
Shaliun_Map: https://www.youtube.com/watch?v=XzzuvGpbHug&list=PL_d_YiRmUA3nnTRmF8QJKyPoEQhrdvpP2&index=6&t=1509s
Borregas_Ave: https://www.youtube.com/watch?v=XSKYE1Oxq-c&list=PL_d_YiRmUA3nnTRmF8QJKyPoEQhrdvpP2&index=12&t=261s
San_Franciso: https://youtu.be/LiSB2RkFow4

**2 - Launch of  Road Occupancy Processor enable /(occupancy_road_status e /gridmap_road_status topics)**

Tutorial: https://autoware.readthedocs.io/en/feature-documentation_rtd/DevelopersGuide/PackagesAPI/semantics/road_occupancy_processor.html

**Attempts/Results;**

San Francisco:
https://studio.youtube.com/video/xZqLfI_NfYs/edit
San Francisco – Debug dos tópicos que falham com Terminal e RQT (Graficamente):
https://youtu.be/mlo_S8NiQgo
Shalun:
https://youtu.be/b9o0L0SxSaI

**3 - Alternative Options**

Tutorial: https://autoware.readthedocs.io/en/feature-documentation_rtd/DevelopersGuide/PackagesAPI/semantics/object_map.html

a) /realtime_cost_map --→ Enabled by  points2costmap node
b) /grid_map_wayarea e /occupancy_wayarea → Enabled by package wararea2grid
c) /filtered_grid_map → Enabled by  grid_map_filter node

**Debug Video of the attempts above:**
https://youtu.be/b9o0L0SxSaI

What is strange is that these costmap generators works with rosbag data or with the first Autoware project from Nayoga university, as the videos below display:

Grid Map Filter
https://www.youtube.com/watch?v=bl-nG8Zv-C0

WayArea To Grid
https://www.youtube.com/watch?v=UkcO4V-0NOw

Road Occupancy Processor
https://www.youtube.com/watch?v=7BLmnC-fb3I

Costmap Generator:
https://www.youtube.com/watch?v=f7kSVJ23Mtw

A summarize of the Result, is that even when the map costmap is produced, it is not updated properly, or even if the obstacle is recognized and the motion planner gets this obstacle input to plan the path properly, the car does not follow the path drawed for the motion planner and stop infinitely behing the obstacle. I would like the car performed an evasive maneuver,but it does not. As you can check in video below, **BETWEEN minute 26 and minute 27:**
https://www.youtube.com/watch?v=XzzuvGpbHug&list=PL_d_YiRmUA3nnTRmF8QJKyPoEQhrdvpP2&index=7&t=1512s

### Steps to reproduce the bug
To reproduce the errors commented above is just follow the videos I have posted on youtube links above. Or follow this tutorial:
https://www.svlsimulator.com/docs/system-under-test/autoware-instructions/.
But before trigger the Localization Module on runtime_manager, check the following packages boxes on runtime_manager:

- On Tab Sensing activate the box ray_ground_filter
- On Tab Computing activate the box costmap_generator 
- 
Finally keep following the tutorial above and launch the LOCALIZATION, MISSION_PLANNING, MOTION_PLANNING TABS to get the result I had.

### Expected behavior

I expected that the Costmap would be created and updated very quickly with robustness. And Afterwards that the MotionPlanner, in this case A*, would get as input the obstacle, draw the trajectory that avoids the obstalce and finally send the proper command velocities to the control to follow this trajectory and perform the evasive maneuever. Actually I expected the Percepetion module and the Planning module of Autoware.AI worked as these modules works very very easily and with robustness in ApolloAuto, as I have done in this simulation:

 https://www.youtube.com/watch?v=bApS6gzyo7s&list=PL_d_YiRmUA3nnTRmF8QJKyPoEQhrdvpP2&index=4&t=253s

### Actual behavior

The topics occupancy_grid_map  or occupancy_road does not generates the map properly. If you try with much effort 10 times it is going to crete the map with the proper obstacles less than 10% of the attempts. And when it does, the Planning module will not send proper controls to allow the car avoid the collision, and the car will stay stopped forever behind the obstacle, even with the evasive manuever already created by the motion planner.

### Screenshots
### Additional Information
Pictures and additional information are described below in ROS Answers:

https://answers.ros.org/question/377659/occupangygrid-map-issue-ray_ground_filter-fails-to-generate-costmap-on-autowareai/

https://answers.ros.org/question/377308/autowareai-perception-failure-agent-not-detected-in-rviz/

https://answers.ros.org/question/377659/occupangygrid-map-issue-ray_ground_filter-fails-to-generate-costmap-on-autowareai/

JWhitleyWork, 2021-06-08:
@marcusvinicius178 Both of your ROS Answers questions were answered at https://answers.ros.org/question/377659/occupangygrid-map-issue-ray_ground_filter-fails-to-generate-costmap-on-autowareai/ and https://answers.ros.org/question/377308/autowareai-perception-failure-agent-not-detected-in-rviz/. Since this is not a confirmed bug, ROS Answers is the correct location to have the discussion. Please continue there, as recommended in several places.

closed 2021-06-08 by JWhitleyWork

marcusvinicius178, 2021-09-01:
Here to send the solution for some users that faced the same issue. 
The tutorials that I have followed and described above in my question are WRONG! If you launch the costmap_generator through the runtime manager or terminal, the occupancy grid map will not be generated or very very rarely....
To fix this there are 2 options:
1). Using Run-time Manager: on Computing Tab click over app option in costmap_generator, scroll down until you find the Lanelet2 checkbox option at the left bottom. CHECK this box ✔️
2) Inside the docker, using a new terminal launch:
`roslaunch costmap_generator costmap_generator_lanelet2.launch`
Ok now the costmap_generator will generate the occupancy_grid_map
3) Just Pay attention that you must check the ray_ground filter or compare_map_filter on Sensing Tab 

However this will not help as the A* keeps not avoiding and crashing the Obstacle, as described in my issue: 

If someone fix this A* please let me know