autowarefoundation/autoware #2384: ndt matching

gky52, 2021-07-13:
a message of over a gigabyte was predicted in tcp ros

gky52, 2021-07-13:
a message of over a gigabyte was predicted in tcp ros

beupon, 2021-08-23:
I make a map(using VLP16, Xsens IMU) and got a pcd file of apartment area. then, I tried to do ndt matching but, problem of mismatching occured. NDT matching worked well on the straight road in APT but mismatching sometimes started when my system turned. I think the reason why parked car in APT which could'nt keep the position all the time and always environment change. how can I handle about this problem?

JWhitleyWork, 2021-08-23:
This is based on the size of the scan. This is not specific to Autoware. However, based on our [Support Guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines), this question should instead be asked on [ROS Answers](https://answers.ros.org) with the `autoware` tag.

closed 2021-08-23 by JWhitleyWork