autowarefoundation/autoware #2388: Using Autoware Docker Containers without Source Code Included

marcusvinicius178, 2021-09-06:
Hi AutowareAI team is the Case 2 tutorial already available from this tutorial website: https://github.com/Autoware-AI/autoware.ai/wiki/Generic-x86-Docker#case-3-creating-a-custom-autoware-docker-container?

[image]

I am trying to use the most recent Open Planner version as **Hatem-Darweesh** suggested me https://github.com/Autoware-AI/core_planning/issues/52 to overcome the collision avoidance issue: https://github.com/Autoware-AI/autoware.ai/issues/2387

However I am struggling to install this from source ( I will probably crash the system as it is said here, in image tutorial below:)
[image]

I decided to work around the DockerFile and set to it get the autoware.ai.repos from a local folder and not from wget. In addition I have changed the dowload of the modules: common, core_planning, and perception modules from master version to open_planner 1.15 branche,:

[image]

and then I have build the docker image with:

`./build.sh
  ./run.sh -t local`

However error raises and then a standard local-cuda-melodic image is created instead my customized with the modules from Hatem-Darshee github repos.

How could I get these forked repo modules (common, planning and percpetion) from Hatem and the other modules from master?

Or at least create a base-container (CASE 2 of tutorial) as I asked in the title of my question?

bahar1372, 2021-09-13:
Hi @marcusvinicius178 . I am looking for a way to modify my Dockerfile like you to accept other repos. could you please tell me how you fixed it at last?

marcusvinicius178, 2021-09-14:
Sure Dude! Just follow my answer here: https://answers.ros.org/question/376512/fork-autoware_ai-repository-and-create-docker-image/

Please ignore the first part (step A) of this answer...Just follow the Step B.
If you have some trouble during the process please let me know.

closed 2021-09-14 by marcusvinicius178

bahar1372, 2021-09-14:
I I face into a trouble since in the autoware.ai repository which I cloned it into a desired folder, there is no install folder to run setup.bash file from it. I mean this part: ~/generic$ source /home/$USERNAME/Autoware/install/local_setup.bash or there is no repository to contain this file: or roslaunch runtime_manager runtime_manager.launch
my desired folder just contain these: 
[image]
 what should I do?

marcusvinicius178, 2021-09-14:
Hey man I think I have also done this....you tried to install AutowareAI CASE2 right? I mean work locally where your modifications should remain....
I can just say that doing this way you will not have success....at least I did not...this is why I opened this issue....
when you build Autoware in a local folder the runtime manager is not found and other modules also not...If you see Autoware team wrote that they would update the tutorial but still did not. That is why I said, just ignore the step A from the link I sent you above....follow the CASE3 of this tutorial: https://github.com/Autoware-AI/autoware.ai/wiki/Generic-x86-Docker#case-3-creating-a-custom-autoware-docker-container
 with the modifications I said on the link I sent you: https://answers.ros.org/question/376512/fork-autoware_ai-repository-and-create-docker-image/

This guy "Hatem Darweesh worked locally with AutowareAI repo as you can check our discussion here: https://github.com/Autoware-AI/core_planning/issues/52#issuecomment-910097419

Just look at his video he does not use Docker: https://www.youtube.com/watch?v=BS5nLtBsXPE

I do not know how he achived that you can just ask him. I prefer to work with Docker in order to do not crash my system...anyway you can try what be best for your case. 
Good Lucky

bahar1372, 2021-09-14:
Thanks @marcusvinicius178 for such a great discussion. just something in your description in  https://answers.ros.org/question/376512/fork-autoware_ai-repository-and-create-docker-image/
make me confused. first when you said using 
./run.sh -b home/desired_empty_folder 
your desired_empty_folder those not contain any folder such as install, autoware, src, build etc because every time I use -b after ./run.sh and set a directory, after running the container, it includes the files which was already on that desired directory. and from every where I cloned autoware.ai github directory, it does not contain install and build folders to let me run autoware software. 
when I just use ./run.sh , I go into a container which include src, build, install, Autoware folders and I have the chance of running autoware software. but if I want to use my desired_empty_folder for ./run.sh command, inside the container I can just access  folders which are inside  desired_empty_folder. 
Second, you said that after running the autoware container with ./run.sh, inside it, use these commands:
~/generic$ cd /home/$USERNAME/Autoware
~/generic$ wget https://raw.githubusercontent.com/Autoware-AI/autoware.ai/1.14.0/autoware.ai.repos
~/generic$ source /home/$USERNAME/Autoware/install/local_setup.bash
~/generic$ vcs import src < autoware.ai.repos
~/generic$ source /opt/ros/$ROS_DISTRO/setup.bash
~/generic$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
~/generic$ source /home/$USERNAME/Autoware/install/local_setup.bash >> /home/$USERNAME/.bashrc

if it is inside the container and after using ./run.sh, why before $ sign, the directory is ~/generic$ so I had a doubt if you are not inside the autoware container while these commands are going be run inside generic folder of the docker before using ./run.sh.
Thanks again for your help to solve this problem.

marcusvinicius178, 2021-09-14:
> Thanks @marcusvinicius178 for such a great discussion. just something in your description in https://answers.ros.org/question/376512/fork-autoware_ai-repository-and-create-docker-image/
> make me [...]
You are right. This is my mistake I was typing the step by step in ROS answers but did not copy from my terminal directly do you understand? Actually just forget there is this Generic folder on these commands...I wrote this tutorial when I was trying to use the Docker with Local folder but as you are facing now issues with this CASE2, I also gave up to use this method. I will repeat again. Try CASE3. As you can check the Autoware.Ai team did not update the CASE2 you wish to use. So just use case 3 as I said, any changes you do in your local folder, you must just push this local folder you have with GitHub, it will be synchronized. Do you know to use git? I can make a tutorial video to show you how using Docker with Autoware and your modifications if you wish... tomorrow maybe...I post here.

And if you take a time to read all the files inside the generic folder ( The docker files and the .sh files), read one by one with calm....as I did...you will understand how the logic works....then you can try all the CASE Autoware provided in your terminal you check what's going on the steps from Docker file that are happen as well as .sh. There are very good, cheap and very quick courses in: https://youtu.be/dJIwrMPwojo
Just do the Docker and the Git Course on your Android it is pretty fast, ( I believe in 1 or 2 days) you finish and get the main concepts.
Finally just try to understand the flags Ater issue:
./run.sh -h
The ./build.sh is the best compilation/build option because it contains the last release code (Master="1.15" version). Later just use ./run.sh -t local
To enter this built image you have done with /. build.sh if you use any other option, such as ./run.sh without flags you will check that other images will be built with other content that will be pulled....this is what the ./run.sh does, it takes a default image that AUTOWARE team decided to let as default ...
The ./run.sh -b /example_directory will result in building the AUTOWARE basic libraries/tools/Environment in this folder....but I really don't know why the other modules are missing...they need to fix or update this tutorial I think. Or there must be additional steps to be done that you can find looking in the internet ( Maybe an old tutorial of old Autoware from Nayoga University before going to Apex.Ai domain, may be lost in some place)
If you use any other ./run.sh -t 1.14 or other release you won't get the updates of master "branch" and won't see the modules fixed with the most updated codes. 
That is what I can say...
> Thanks again for your help to solve this problem.

baharkhabbazan, 2021-09-16:
Hi @marcusvinicius178 . Did you succeed to make a tutorial video? It will be so helpful for somebody like me.

marcusvinicius178, 2021-09-20:
Hi @baharkhabbazan yes I did it this weekend:

https://youtu.be/ewlMbr_XrEA

Good Lucky!

hatem-darweesh, 2021-09-20:
@marcusvinicius178 good job with the tutorial video, 
can you make it public and "not for kids" so I can share it. 

Regards,

baharkhabbazan, 2021-09-20:
Hi @marcusvinicius178  I did exactly what you have done on this tutorial and it went well. Thanks a lot for helping me to solve this issue after several weeks.

marcusvinicius178, 2021-09-23:
Hi @baharkhabbazan glad to hear. 
Thanks Mr. @hatem-darweesh sorry for the delay I am working hard on my thesis. I have changed, just check if you can share it now please.

hatem-darweesh, 2021-09-23:
Great, it works, good luck with your thesis.

mshah72, 2023-01-24:
How to restart Autoware every time if it is build inside Docker (created using the steps mentioned in [this](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/).

Please explain step-by-step.