autowarefoundation/autoware #2392: Op flag is not working with op_planner (autoware.ai 1.14)

jiamingZhong93, 2021-11-07:
These are the nodes I activated:

- vel_pose_connect
- op_global_planner
- pure_pursuit
- twist_filter
- wf_simulatior
- op_common_params
- op_trajectory_generator
- op_motion_predictor
- op_trajectory_evaluator
- op_behavior_selector
- op_perception_simulator
- lidar_kf_contour_track

I can run the planning and control simulation and show in RVIZ.
However, the Op Flag just didn't work when I tried to add obstacle in the RVIZ. 
[image]

hatem-darweesh, 2021-11-08:
Op flag has nothing to do with OpenPlanner. 
To insert static object to use with op_perception_simulator use "2D Nav Goal".

jiamingZhong93, 2021-11-10:
> Op flag has nothing to do with OpenPlanner. To insert static object to use with op_perception_simulator use "2D Nav Goal".

Thank you Hatem for quick comment!

As you suggested, I run the op_perception_simulator and a new button Plant_flag appearred on the RVIZ. And, this time I could pin the Plant_flag (blue circle) on the map. 

However, these Plant_flags did not function as an obstacles. The robot just drives directly across.
I have tried differend op_plan parameters (parameters in the pic bellow might not be appropriate) and the robot always drive through. 
Is there anything I did not operate correctly?
[image]

By the way, I guess the "2D Nav Goal" is only used for setting destination, and you actually mean "Plant_flag", right?

Thanks again!

hatem-darweesh, 2021-11-10:
No I really mean "2D Nav Goal"! 
I neve use "Plant_flag" or "Op flag" in my life.

hatem-darweesh, 2021-11-10:
also check this question: 
https://answers.ros.org/question/390380/autoware-hatem-1130-object-avoidance-not-working/

jiamingZhong93, 2021-11-14:
> also check this question: https://answers.ros.org/question/390380/autoware-hatem-1130-object-avoidance-not-working/

Thank you so much Hatem! This issue close! Thank you!

closed 2021-11-14 by jiamingZhong93