autowarefoundation/autoware #2394: vision_darknet_detect node died: CUDA Error:invalid device symbol

xiewenjing1170, 2021-11-22:
## Bug report

### Required information:

- Operating system and version:
  - Ubuntu 18.04
  - GPU: NVIDIA GeForce RTX 3080 (UUID: GPU-f353cfb1-6d90-c11a-df7c-7cfe4e29a28f)
  - NVIDIA driver: 470.82.00
  - CUDA: 10.0.130
  - Qt 5.12.8
  - Eigen 3.3.7
- Autoware installation type:
  - source build. Ref to the official tutorial: [autoware.ai source build](https://github.com/Autoware-AI/autoware.ai/wiki/Source-Build)
- Autoware version or commit hash
  - Autoware.ai 1.14
- ROS distribution and version: 
  - ROS version:Melodic
- ROS installation type:
  - official tutorial, Ref: [](http://wiki.ros.org/Installation/Ubuntu)
- Package or library, if applicable:
  - 
 

### Description of the bug
When I run runtime_manager.launch and load Yolov3. The vision_darket_detect node dies all the time. The error as follows:

CUDA Error: invalid device symbol: File exists
Failed to find match for field 'intensity'.
[vision_darknet_detect-1] process has died [pid 6308, exit code 255, cmd /home/wenjing/autoware.ai/install/vision_darknet_detect/lib/vision_darknet_detect/vision_darknet_detect __name:=vision_darknet_detect __log:=/home/wenjing/.ros/log/c2a06674-4ade-11ec-b1b1-a4bb6d69b188/vision_darknet_detect-1.log].
log file: /home/wenjing/.ros/log/c2a06674-4ade-11ec-b1b1-a4bb6d69b188/vision_darknet_detect-1*.log

+ When nodes running, the memory is enough.
I have no idea about this CUDA Error. I will really appreciate if anyone can help me. (I have tried several methods, but they doesn't work.)

### Steps to reproduce the bug

### Expected behavior

### Actual behavior

### Screenshots

### Additional information

bruce-almon, 2021-12-15:
Hello, try docker build instead.
The reason is that you are using CUDA 10.0 on a 3080
30 series GPU is **NOT** compatible with CUDA <11.0
You should install at least CUDA 11.0  devel lib for your 3080.
I can run the docker version on my 3090.

closed 2022-06-24 by mitsudome-r

wzr6009, 2022-08-27:
Hello, after I build with docker, cuda is 11.1 but CUDA Error: invalid device symbol will still appear.

yinwei-zhang, 2023-02-12:
Hi,
Did the problem get solved? I am having the same error, but I don't know how to solve it.

wenjing1170, 2023-02-12:
@yinwei-zhang My error was caused by GPU RTX3080. 30 series GPU is NOT compatible with CUDA <11.0. In the end, I switched to a machine with GPU 2080 to run the program successfully!