autowarefoundation/autoware #2395: Invoking "make -j8 -l8" failed

Zong-Ming-Jing, 2021-12-28:
i have a problem like this.
[image]

I don't know how to fix. HELP~!

my workplace is: Ubuntu 16.04
                            ROS-kinetic
                            QT 5.9.1
                            CUDA 9.0 (GPU 2080 Ti 11G)
                            cudnn 7.4.2
                            Autoware 1.10.0

I really need the help. Help me! Please! 
thanks!

urbansound8K, 2022-02-09:
did you fix it?
I got the sam
[image]
[image]
[image]
e

urbansound8K, 2022-05-10:
fixed 
checked the compatible of ur GPU with the requirements.

closed 2022-06-24 by mitsudome-r