autowarefoundation/autoware #2396: Compile error when build with Driveworks on Drive PX2

Torreskai0722, 2022-01-01:
We have build autoware.ai on Drive PX2 with Drivework 1.2.4. However, the package autoware_driveworks_gmsl_interface, autoware_driveworks_interface has got complie errors. We cannot find dwImageFormatConverterHandle_t either on the Drive PX2 or the host machine. How can we find it and build driveworks wth autoware?

OS: Ubuntu 16.04
CUDA: 9.2
Driveworks version:  1.2.4
Autoware.ai version: 1.12.0

```
nvidia@tegra-ubuntu:~/autoware.ai$ AUTOWARE_COMPILE_WITH_CUDA=1 colcon build --symlink-install --packages-select autoware_driveworks_gmsl_interface autoware_driveworks_interface --cmake-args -DCMAKE_BUILD_TYPE=Release --cmake-clean-cache
Starting >>> autoware_driveworks_gmsl_interface
Starting >>> autoware_driveworks_interface                       
--- stderr: autoware_driveworks_interface                                                                                                             
9.2
aarch64
/usr/local/driveworks/lib/libdriveworks.so
TRUE
/usr/local/cuda-9.2
---
Finished <<< autoware_driveworks_interface [20.1s]
--- stderr: autoware_driveworks_gmsl_interface                               
CMake Warning (dev) at CMakeLists.txt:68 (target_link_libraries):
  Policy CMP0023 is not set: Plain and keyword target_link_libraries
  signatures cannot be mixed.  Run "cmake --help-policy CMP0023" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  The plain signature for target_link_libraries has already been used with
  the target "autoware_driveworks_gmsl_interface_node".  All uses of
  target_link_libraries with a target should be either all-keyword or
  all-plain.

  The uses of the plain signature are here:

   * /usr/share/cmake-3.5/Modules/FindCUDA.cmake:1704 (target_link_libraries)

This warning is for project developers.  Use -Wno-dev to suppress it.

In file included from /home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/SekonixGmslCamera.hpp:71:0,
                 from /home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/src/gmsl_camera_multiple/SekonixGmslCamera.cpp:37:
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/DriveWorksApi.hpp:95:3: error: ‘dwImageFormatConverterHandle_t’ does not name a type
   dwImageFormatConverterHandle_t yuv2rgba;
   ^
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/DriveWorksApi.hpp:142:44: error: ‘dwImageFormatConverterHandle_t’ has not been declared
                          uint32_t sibling, dwImageFormatConverterHandle_t yuv2rgba,
                                            ^
In file included from /home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/SekonixGmslCamera.hpp:71:0,
                 from /home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/src/gmsl_camera_multiple/gmsl_nodelets.cpp:43:
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/DriveWorksApi.hpp:95:3: error: ‘dwImageFormatConverterHandle_t’ does not name a type
   dwImageFormatConverterHandle_t yuv2rgba;
   ^
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/DriveWorksApi.hpp:142:44: error: ‘dwImageFormatConverterHandle_t’ has not been declared
                          uint32_t sibling, dwImageFormatConverterHandle_t yuv2rgba,
                                            ^
In file included from /home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/SekonixGmslCamera.hpp:71:0,
                 from /home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/src/gmsl_camera_multiple/gmsl_node.cpp:40:
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/DriveWorksApi.hpp:95:3: error: ‘dwImageFormatConverterHandle_t’ does not name a type
   dwImageFormatConverterHandle_t yuv2rgba;
   ^
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/DriveWorksApi.hpp:142:44: error: ‘dwImageFormatConverterHandle_t’ has not been declared
                          uint32_t sibling, dwImageFormatConverterHandle_t yuv2rgba,
                                            ^
make[2]: *** [CMakeFiles/nodelet_autoware_driveworks_gmsl_interface.dir/src/gmsl_camera_multiple/SekonixGmslCamera.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/SekonixGmslCamera.hpp:71:0,
                 from /home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/src/gmsl_camera_multiple/SekonixGmslCamera.cpp:37:
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/DriveWorksApi.hpp:95:3: error: ‘dwImageFormatConverterHandle_t’ does not name a type
   dwImageFormatConverterHandle_t yuv2rgba;
   ^
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/include/DriveWorksApi.hpp:142:44: error: ‘dwImageFormatConverterHandle_t’ has not been declared
                          uint32_t sibling, dwImageFormatConverterHandle_t yuv2rgba,
[... 129 lines cut ...]
          ^
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/src/gmsl_camera_multiple/DriveWorksApi.cpp: In member function ‘void DriveWorks::DriveWorksApi::releaseCameras(DriveWorks::Camera*)’:
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/src/gmsl_camera_multiple/DriveWorksApi.cpp:525:51: error: ‘struct DriveWorks::Camera’ has no member named ‘yuv2rgba’
     dwImageFormatConverter_release(&cameraSensor->yuv2rgba);
                                                   ^
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/src/gmsl_camera_multiple/DriveWorksApi.cpp:525:59: error: ‘dwImageFormatConverter_release’ was not declared in this scope
     dwImageFormatConverter_release(&cameraSensor->yuv2rgba);
                                                           ^
/home/nvidia/autoware.ai/src/drivers/awf_drivers/autoware_driveworks_gmsl_interface/src/gmsl_camera_multiple/DriveWorksApi.cpp:532:51: error: ‘dwImageNvMedia_destroy’ was not declared in this scope
    dwStatus result = dwImageNvMedia_destroy(&frame);
                                                   ^
make[2]: *** [CMakeFiles/autoware_driveworks_gmsl_interface_node.dir/src/gmsl_camera_multiple/SekonixGmslCamera.cpp.o] Error 1
make[2]: *** [CMakeFiles/nodelet_autoware_driveworks_gmsl_interface.dir/src/gmsl_camera_multiple/DriveWorksApi.cpp.o] Error 1
make[1]: *** [CMakeFiles/nodelet_autoware_driveworks_gmsl_interface.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/autoware_driveworks_gmsl_interface_node.dir/src/gmsl_camera_multiple/DriveWorksApi.cpp.o] Error 1
make[1]: *** [CMakeFiles/autoware_driveworks_gmsl_interface_node.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< autoware_driveworks_gmsl_interface [22.5s, exited with code 2]

Summary: 1 package finished [24.8s]
  1 package failed: autoware_driveworks_gmsl_interface
  2 packages had stderr output: autoware_driveworks_gmsl_interface autoware_driveworks_interface
```

TheGermanKid, 2022-01-08:
not sure if this is the same issue but looks similiar to this issue : - [https://forums.developer.nvidia.com/t/fail-to-install-autoware-with-driveworks-on-drive-px2/167148](url)

Torreskai0722, 2022-01-12:
> not sure if this is the same issue but looks similiar to this issue : - [https://forums.developer.nvidia.com/t/fail-to-install-autoware-with-driveworks-on-drive-px2/167148](url)

This issue is different with the issue. We cannot find the definition of ```dwImageFormatConverterHandle_t``` either on the Drive PX2 or the host machine.